WebODM/WebODM #323: Hard to understand installation directions

dalearn, 2017-11-12:
As someone unused to docker, can you update the installation directions please?  Do I need to set up a new docker container first?  If so how would I go about doing that on Linux and after that, how would I run this program?  I just want to stitch together my images using this awesome program

pierotofy, 2017-11-12:
Hi @dalearn, have you followed the README? What parts are not clear?

dalearn, 2017-11-12:
I downloaded docker and after much struggle finally got the config file to run but it appears to be downloading hundreds of files.  What are these files for?

jupeuav, 2017-11-12:
I am also having issues with the install. I followed the instructions got 'Getting Started'. Entered the command into terminal and everything seems to run fine until I get to the below:

Starting webapp     ... error

dalearn, 2017-11-12:
make sure you log out and log back in...   I saw "ERROR: No containers to start" in there somewhere but now all I see is a bunch of downloads.

dalearn, 2017-11-12:
It has almost entirely used up my disk storage and is still going!

closed 2017-11-18 by pierotofy

dalearn, 2017-11-18:
I finally figured out the problem, I was getting the error: `standard_init_linux.go:195: exec user process caused "exec format error"` which appears that I was using an X86 image.  How could I obtain an ARM image?

pierotofy, 2017-11-18:
@dalearn your first step would be to compile OpenDroneMap from sources. I'm not sure this has been attempted before, but could be worth a try. Unfortunately I don't have precise guidelines on how to do this, start from https://github.com/OpenDroneMap/OpenDroneMap#diving-deeper and see if you can make it work.

pierotofy, 2017-11-18:
Might also be worth to open a discussion on http://community.opendronemap.org to see if anybody else has done it before.

dalearn, 2017-11-18:
How would I compile it from sources?  After I pull it from Github, what do I do?  Where do I go to compile it?

dalearn, 2017-11-18:
As a web developer who usually stays within the confines of my IDE, I am trying to run this on a Raspberry Pi and I just want it to run instead of having to install all sorts of things.  I am going to a high profile conference soon and would like to use this and promote this because it is open source.

pierotofy, 2017-11-18:
Mm, I would discourage using a Raspberry PI, I doubt you'll have enough memory to process any meaningful dataset size. Use a build similar to this if you're looking for portability: http://posm.io/getting-a-posm/

dalearn, 2017-11-18:
could I chain two raspberry pis or use a mac?

pierotofy, 2017-11-18:
You could use a Mac (if it's Intel-based). You cannot chain two raspberry pis.

dalearn, 2017-11-18:
Would 8GB of RAM be enough?  How would I install it on MacOS without using the terminal a bunch?

pierotofy, 2017-11-18:
Yes 8GB will get you up and running. Just follow the instructions here: https://github.com/OpenDroneMap/WebODM#getting-started

dalearn, 2017-11-18:
It says something about node-opendronemap, Could I use an army of Pis running that or would it again be too big?

pierotofy, 2017-11-18:
Unfortunately no, you couldn't use an array of PIs. The processing is too memory intensive.