WebODM/WebODM #328: reset password of webodm

green512, 2017-11-16:
I forgot the password of the wenodm. How can I reset the password?

pierotofy, 2017-11-16:
Hi @green512,

It looks like there is some information missing from your ticket that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your ticket has to contain in order to be processable.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2017-11-22 22:40) I'll close this ticket so it doesn't clutter the bug tracker.

Best regards,
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

green512, 2017-11-16:
I use cmd "./webodm.sh resetadminpassword" to reslove the problem.

pierotofy, 2017-11-24:
Since apparently some of the required information is still missing, I'm closing this now, sorry. Feel free to reopen this or create a new issue once you can provide **all** required information.

closed 2017-11-24 by pierotofy