WebODM/WebODM #330: Process exited with code 132

mxcoldhit, 2017-11-16:
Installed with Docker.
Using Chrome with windows 10 Pro 64bit
I have tried different versions of Python. I have used most recent and have downgraded to 3.5 64bit as of right now.

When trying to run my first set of photos, i get he following error:
/code/run.sh: line 6:    19 Illegal instruction     python $RUNPATH/run.py "$@"

Im not sure if it has to do with the environment path or not. I have checked the paths. Everything is set up correctly to my knowledge, but im no genius by any means.

I am using the default settings.

Since I couldnt find any issues like this, i will assume its my noob mistake somewhere.

Any help would be greatly appreciated. Thank you

pierotofy, 2017-11-16:
Hi @mxcoldhit,

It looks like there is some information missing from your ticket that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your ticket has to contain in order to be processable.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2017-11-23 15:50) I'll close this ticket so it doesn't clutter the bug tracker.

Best regards,
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2017-11-17:
Hi @mxcoldhit, you are probably using an older CPU. The docker setup needs CPUs with 64-bit extensions, MMX, SSE, SSE2, SSE3 and SSSE3 instruction set support or higher.

You could rebuild the OpenDroneMap and node-OpenDroneMap docker images from sources and then use that instance of node-OpenDroneMap to process a task.

closed 2017-12-22 by pierotofy

pierotofy, 2018-07-18:
Hey @julianpalladino :hand: , no, that will only rebuild the WebODM containers.

You need to rebuild ODM (https://github.com/OpenDroneMap/OpenDroneMap), then node-ODM (https://github.com/OpenDroneMap/node-OpenDroneMap), then finally you can use the node-ODM image with WebODM (no need to recompile WebODM's containers). Usually a:

`docker build -t <appropriate tag name> .`

Will do the trick for each repository.

julianpalladino, 2018-07-26:
Hi again @pierotofy  , thank you very much for your help!
I tried building OpenDroneMap and node-OpenDroneMap natively. However I still have the same problem (code 132 / illegal instruction, due to old CPU), exactly as [this issue](https://github.com/OpenDroneMap/OpenDroneMap/issues/859).

Is it a bug? Is there a way to bypass that part of the program? I wouldn't mind to lose some functionality, as long as I can use the rest of the program

pierotofy, 2018-07-31:
Hey @julianpalladino this is a great find! Would you mind sharing more about the process you followed to remove the flag? It would be great to document the fix for other users (or even open a PR).

fragofer, 2018-08-08:
Hi @pierotofy, we're working with @julianpalladino on this issue. 
The problem wasn't in ODM, node-ODM nor WebODM so rebuilding these projects led us to the same error. After long hours of debugging we found that the problem was/is in a related project MVE (used by mvs-texturing): https://github.com/nmoehrle/mve 

In particular, in the file:
https://github.com/simonfuhrmann/mve/blob/master/Makefile.inc
They added the flag -sse4 which is not supported by old AMD processors (ours is an AMD Phenom II X6 1090T). So, we removed that flag and now everything is working :-)

I think that the best thing to do is to transform MVE in a CMakeList based project, so for now a good fix would be to change:
https://github.com/OpenDroneMap/mvs-texturing/blob/master/elibs/CMakeLists.txt
and point to forked projects that can be controlled by the ODM team, and also try to avoid cloning external libraries from master branches to gain stability in the building process.

Thanks!
Francisco

fpolig01, 2020-03-31:
I'm having a similar issue with AMD Phenom(tm) II X6 1090T CPU. I just started working with ODM. Can this CPU be used? I'm unfamiliar with how to recompile, but will investigate if it is possible to get this CPU working.