WebODM/WebODM #331: leak somewhere in webodm/gunicorn/wsgi processing

pee, 2017-11-17:
### How did you install WebODM? (Docker, natively, ...)?
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Win10/Chrome  Linux/Chrome

### What is the problem?
Apparent memory leak in webodm gunicorn wsgi process. Largish 700 pic batch process fails with out of memory, restarting docker container and rerun direct on node-xx docker container passes.

### What should be the expected behavior?
It should finish

Quick container memory snapshot after failed run.
,,,
root@87cef3ac7304:/webodm# ps auxwwwwm
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  0.0  0.0  21752     4 ?        -    Nov16   0:00 /bin/bash -c chmod +x /webodm/*.sh && /bin/bash -c "/webodm/wait-for-it.sh db:5432 -- /webodm/start.sh --create-default-pnode"
root         -  0.0    -      -     - -        Ss   Nov16   0:00 -
root         8  0.0  0.0  21772     8 ?        -    Nov16   0:00 /bin/bash /webodm/start.sh --create-default-pnode
root         -  0.0    -      -     - -        S    Nov16   0:00 -
root        39  0.0  0.0  32388     4 ?        -    Nov16   0:00 nginx: master process nginx -c /webodm/nginx/nginx.conf
root         -  0.0    -      -     - -        Ss   Nov16   0:00 -
root        40  0.0  0.0  32848   320 ?        -    Nov16   0:04 nginx: worker process
root         -  0.0    -      -     - -        S    Nov16   0:04 -
root        41  0.0  0.0 517536  1556 ?        -    Nov16   0:06 /usr/local/bin/python /usr/local/bin/gunicorn webodm.wsgi --bind unix:/tmp/gunicorn.sock --timeout 360 --preload
root         -  0.0    -      -     - -        S    Nov16   0:06 -
root        47  0.8 19.5 17341708 6400236 ?    -    Nov16   8:48 /usr/local/bin/python /usr/local/bin/gunicorn webodm.wsgi --bind unix:/tmp/gunicorn.sock --timeout 360 --preload
root         -  0.3    -      -     - -        Sl   Nov16   3:40 -
root         -  0.0    -      -     - -        Sl   Nov16   0:04 -
root         -  0.0    -      -     - -        Sl   Nov16   0:13 -
root         -  0.0    -      -     - -        Sl   Nov16   0:03 -
root         -  0.0    -      -     - -        Sl   Nov16   0:13 -
root         -  0.0    -      -     - -        Sl   Nov16   0:13 -
root         -  0.0    -      -     - -        Sl   Nov16   0:13 -
root         -  0.0    -      -     - -        Sl   Nov16   0:12 -
root         -  0.0    -      -     - -        Sl   Nov16   0:13 -
root         -  0.0    -      -     - -        Sl   Nov16   0:12 -
root         -  0.0    -      -     - -        Sl   Nov16   0:12 -
root         -  0.0    -      -     - -        Sl   Nov16   0:13 -
root      4530  0.0  0.0  21968  2692 pts/0    -    Nov16   0:00 /bin/bash
root         -  0.0    -      -     - -        Ss   Nov16   0:00 -
root     12646  0.0  0.0  19192  2400 pts/0    -    01:18   0:00 ps auxwwwwm
root         -  0.0    -      -     - -        R+   01:18   0:00 -

While the job is running you can check pmap, the number of 65k? anon block allocations continues to grow while running, they are not freed. When i've seen this previously it was related to thread mmap() leak issues, but i'm not familiar with gunicorn.

root@87cef3ac7304:/webodm# pmap 47 | grep anon | wc
    617    3702   25914
,,,

Working on learning how to debug gunicorn ;p, but hopefully this gives you a heads up.

pierotofy, 2017-11-17:
Hey @pee thanks for reporting this. The only major thing that I can think of that could be taking a lot of memory is the file upload mechanism from webodm to node-odm https://github.com/OpenDroneMap/WebODM/blob/master/nodeodm/api_client.py#L70. All images currently need to be loaded in memory and sent to the processing node via a POST request.

That code is called in a background thread via https://github.com/OpenDroneMap/WebODM/blob/master/app/scheduler.py#L52

The first step would be to understand if this is just a gunicorn problem or if this happens on the django dev server as well (run with: `./devenv.sh start` instead of `./webodm.sh start`).

pee, 2017-11-17:
comedy of errors begins, I'll keep poking at it

(Output from ./devenv.sh start)
,,,
webapp        | Checking python version... 3.x, good!
webapp        | Checking GDAL version... GDAL 2.2.2, released 2017/09/15, excellent!
webapp        | Setup git modules...
webapp        | Cloning into 'nodeodm/external/node-OpenDroneMap'...
webapp        | fatal: unable to access 'https://github.com/OpenDroneMap/node-OpenDroneMap/': Could not resolve host: github.com
webapp        | Clone of 'https://github.com/OpenDroneMap/node-OpenDroneMap' into submodule path 'nodeodm/external/node-OpenDroneMap' failed
webapp        | Setup npm dependencies...
webapp        | npm WARN deprecated tween.js@16.6.0: This package has moved to @tweenjs/tween.js. Please update your dependencies.
webapp        | npm WARN deprecated babel-preset-es2015@6.24.1: 🙌  Thanks for using Babel: we recommend using babel-preset-env now: please read babeljs.io/env to update!
,,,

Meanwhile, I'm running just the upload part in another session right now, it will take a bit, i'll post anything that comes out of it.

pee, 2017-11-17:
Memory stats from just built docker image, 700 files uploaded

[cleanrun.txt](https://github.com/OpenDroneMap/WebODM/files/1481190/cleanrun.txt)

root@b00858a4afc4:/webodm# du -sk app/media/
3584944 app/media/

So with no processing started, and no files transferred from the webapp to the processing node, we're ok. Looks like your idea was good. I'll see if i can get a trace on the container when i hit the start processing button.

pee, 2017-11-17:
Hacked up a quick script to check trace of the container, found nothing related to file passing however there are quite a few points like this (db access?) that seem to cause vm growth. I'm still learning the code base, so I'm posting now in case someone with it pre loaded in their brain wants to take a look.
,,,
24896323 14:15:46.603844434 4 gunicorn (12784) < sendto res=210 data=Q....SELECT "app_project"."id", "app_project"."owner_id", "app_project"."name",
24896324 14:15:46.603845599 4 gunicorn (12784) > poll fds=13:411 timeout=4294967295
24896325 14:15:46.603847894 4 gunicorn (12784) > switch next=0 pgft_maj=0 pgft_min=237265 vm_size=6841068 vm_rss=5616260 vm_swap=652
24896326 14:15:46.604009613 4 gunicorn (12784) < poll res=1 fds=13:41
24896327 14:15:46.604011053 4 gunicorn (12784) > recvfrom fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=2097152
24896328 14:15:46.604012249 4 gunicorn (12784) < recvfrom res=260 data=T......id...F...............owner_id...F...............name...F...............de tuple=NULL
24896329 14:15:46.604196661 4 gunicorn (12784) > write fd=2(<p>pipe:[6591863]) size=65
24896330 14:15:46.604202434 4 gunicorn (12784) < write res=65 data=INFO Acquiring lock: Task [Task of 2017-11-17T09:49:17.935Z] (2).
24896331 14:15:46.604546658 4 gunicorn (12784) > sendto fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=79 tuple=NULL
24896332 14:15:46.604561093 4 gunicorn (12784) < sendto res=79 data=Q...NSELECT (1) AS "a" FROM "app_project" WHERE "app_project"."id" = 1 LIMIT 1.
24896333 14:15:46.604562261 4 gunicorn (12784) > poll fds=13:411 timeout=4294967295
24896334 14:15:46.604564058 4 gunicorn (12784) > switch next=0 pgft_maj=0 pgft_min=237265 vm_size=6841068 vm_rss=5616260 vm_swap=652
24896335 14:15:46.604698347 4 gunicorn (12784) < poll res=1 fds=13:41
24896336 14:15:46.604699787 4 gunicorn (12784) > recvfrom fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=2097152
24896337 14:15:46.604700981 4 gunicorn (12784) < recvfrom res=59 data=T......a...................D..........1C....SELECT 1.Z....I tuple=NULL
24896338 14:15:46.605013456 4 gunicorn (12784) > sendto fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=101 tuple=NULL
24896339 14:15:46.605027555 4 gunicorn (12784) < sendto res=101 data=Q...dSELECT (1) AS "a" FROM "nodeodm_processingnode" WHERE "nodeodm_processingno
24896340 14:15:46.605028716 4 gunicorn (12784) > poll fds=13:411 timeout=4294967295
24896341 14:15:46.605030724 4 gunicorn (12784) > switch next=0 pgft_maj=0 pgft_min=237265 vm_size=6841068 vm_rss=5616260 vm_swap=652
24896342 14:15:46.605153256 4 gunicorn (12784) < poll res=1 fds=13:41
24896343 14:15:46.605154661 4 gunicorn (12784) > recvfrom fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=2097152
24896344 14:15:46.605155851 4 gunicorn (12784) < recvfrom res=59 data=T......a...................D..........1C....SELECT 1.Z....I tuple=NULL
24896345 14:15:46.605748326 4 gunicorn (12784) > mmap addr=0 length=134217728 prot=0(PROT_NONE) flags=74(MAP_PRIVATE|MAP_ANONYMOUS|MAP_NORESERVE) fd=4294967295 offset=0
24896346 14:15:46.605753998 4 gunicorn (12784) < mmap res=7FF7BC000000 vm_size=6972140 vm_rss=5616260 vm_swap=652
24896347 14:15:46.605754418 4 gunicorn (12784) > munmap addr=7FF7C0000000 length=67108864
24896348 14:15:46.605758060 4 gunicorn (12784) < munmap res=0 vm_size=6906604 vm_rss=5616260 vm_swap=652
24896349 14:15:46.605758412 4 gunicorn (12784) > mprotect
24896350 14:15:46.605759455 4 gunicorn (12784) < mprotect
24896351 14:15:46.607051158 4 gunicorn (12784) > mprotect
24896352 14:15:46.607052873 4 gunicorn (12784) < mprotect
24896353 14:15:46.607353044 4 gunicorn (12784) > sendto fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=11 tuple=NULL
24896354 14:15:46.607369685 4 gunicorn (12784) < sendto res=11 data=Q....BEGIN.
24896355 14:15:46.607371015 4 gunicorn (12784) > poll fds=13:411 timeout=4294967295
24896356 14:15:46.607373279 4 gunicorn (12784) > switch next=0 pgft_maj=0 pgft_min=237868 vm_size=6906604 vm_rss=5618636 vm_swap=652
24896357 14:15:46.607455658 4 gunicorn (12784) < poll res=1 fds=13:41
24896358 14:15:46.607457167 4 gunicorn (12784) > recvfrom fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=2097152
24896359 14:15:46.607458374 4 gunicorn (12784) < recvfrom res=17 data=C....BEGIN.Z....T tuple=NULL
24896360 14:15:46.607566966 4 gunicorn (12784) > sendto fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=1228800 tuple=NULL
24896361 14:15:46.607787554 4 gunicorn (12784) < sendto res=1228800 data=Q...GUPDATE "app_task" SET "uuid" = 'f7284e25-543d-488f-b94d-80dba287dbfa', "pro
24896362 14:15:46.607907362 4 gunicorn (12784) > sendto fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=4168 tuple=NULL
24896363 14:15:46.607914412 4 gunicorn (12784) < sendto res=4168 data=...100 - done..0...10...20...30...40...50...60...70.10...20...30...40...50...60.
24896364 14:15:46.607915244 4 gunicorn (12784) > poll fds=13:411 timeout=4294967295
24896365 14:15:46.607917158 4 gunicorn (12784) > switch next=0 pgft_maj=0 pgft_min=237869 vm_size=6906604 vm_rss=5618636 vm_swap=652
24896366 14:15:46.624351976 4 gunicorn (12784) < poll res=1 fds=13:41
24896367 14:15:46.624354990 4 gunicorn (12784) > recvfrom fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=2097152
24896368 14:15:46.624356331 4 gunicorn (12784) < recvfrom res=20 data=C....UPDATE 1.Z....T tuple=NULL
24896369 14:15:46.624371208 4 gunicorn (12784) > madvise
24896370 14:15:46.624411956 4 gunicorn (12784) < madvise
24896371 14:15:46.624758394 4 gunicorn (12784) > sendto fd=13(<4t>172.18.0.4:60950->172.18.0.3:5432) size=12 tuple=NULL
,,,

pierotofy, 2017-11-17:
If the leak is caused by DB access, then we should be able to observe this without any task processing happening. This could be caused by the scheduler not releasing the database connection? https://github.com/OpenDroneMap/WebODM/blob/master/app/scheduler.py

pierotofy, 2017-11-17:
Possibly relevant: https://github.com/agronholm/apscheduler/issues/223

pee, 2017-11-17:
Could be, but the rss increases as well, and memory pressure does not cause a free

pee, 2017-11-19:
Got the dev setup to work(my dns server was angry about 172.x.x.x clients). Looks the same.
,,,
root@5b272c76bc16:/webodm# ps auxwwww
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  0.0  0.0  21756     4 ?        Ss   Nov18   0:00 /bin/bash -c chmod +x /webodm/*.sh && /bin/bash -c "/webodm/wait-for-it.sh db:5432 -- /webodm/start.sh --create-default-pnode --setup-devenv"
root         8  0.0  0.0  21772   116 ?        S    Nov18   0:00 /bin/bash /webodm/start.sh --create-default-pnode --setup-devenv
root       137  0.1  1.6 1659620 523272 ?      Sl   Nov18   0:21 node /usr/bin/webpack --watch
root       161  0.0  0.1 507572 37048 ?        S    Nov18   0:00 python manage.py runserver 0.0.0.0:8000
root       166  3.3 10.3 4670108 3282856 ?     Sl   Nov18   8:16 /usr/local/bin/python manage.py runserver 0.0.0.0:8000
,,,
Could use some suggestions on instrumenting for db debug.

pierotofy, 2017-11-19:
Does `RSS` increase when observing the `python manage.py runserver` process? If not, perhaps this is specific to gunicorn.

pee, 2017-11-19:
It shows the same growth pattern with the same trace.
```
root@5b272c76bc16:/webodm# ps auxwww
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  0.0  0.0  21756     4 ?        Ss   Nov18   0:00 /bin/bash -c chmod +x /webodm/*.sh && /bin/bash -c "/webodm/wait-for-it.sh db:5432 -- /webodm/start.sh --create-default-pnode --setup-devenv"
root         8  0.0  0.0  21772     8 ?        S    Nov18   0:00 /bin/bash /webodm/start.sh --create-default-pnode --setup-devenv
root       137  0.0  0.0 1659620    4 ?        Sl   Nov18   0:21 node /usr/bin/webpack --watch
root       161  0.0  0.0 507572    16 ?        S    Nov18   0:00 python manage.py runserver 0.0.0.0:8000
root       166  1.9  9.4 5269660 3002952 ?     Sl   Nov18  25:51 /usr/local/bin/python manage.py runserver 0.0.0.0:8000
```

pierotofy, 2017-11-20:
I can reproduce this while using gunicorn, a workaround could be to run gunicorn with a --max-requests parameter (https://rayed.com/wordpress/?p=1669) which will restart the worker after a while (not an ideal solution).

A good source for debugging more in depth: http://chase-seibert.github.io/blog/2013/08/03/diagnosing-memory-leaks-python.html

pierotofy, 2017-11-20:
Also part of the problem is the Debug=True flag set in webodm/settings.py by default. https://docs.djangoproject.com/en/dev/faq/models/#why-is-django-leaking-memory

pee, 2017-11-20:
That certainly wouldn't help. Right now each Task instantiation hits the db for about 1.2meg+ of console_output, unless it's pulling from a cached object that I haven't found yet.

pierotofy, 2020-04-02:
Lots has changed in WebODM since this was first opened. If this is still an issue, submit a PR? :+1:

closed 2020-04-02 by pierotofy