WebODM/WebODM #333: Process exited with code 1

journeyman44, 2017-11-19:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 62 / Windows 10

### What is the problem?

File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

### What should be the expected behavior?

Completed map / object

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Uploaded 170 images for test data from Yuneec H520.

If I use use-pmvs true option it completes, but looks poor in comparison to the likes of Drone Deploy and even Maps Made Easy (which is usually rubbish). Even MS's own new ICE app produces better results.

Upped min-num-features to 12000 and still no joy.

Trying to get the best output to evaluate this tool as an alternative for orthos and 3d models, etc from my drone work.

VM has 8gb memory and 100gb ssd space allocated.

Attached the task output.

Have images in Google drive if required.

Thanks for any help anyone can give.

[task_output.txt](https://github.com/OpenDroneMap/WebODM/files/1485292/task_output.txt)

pierotofy, 2017-11-19:
Hi @journeyman44, thanks for the detailed report.

Per the task_output.txt, `std::bad_alloc` signifies that the program ran out of memory while creating the mesh output. We should probably have a better error message for that, I opened an improvement issue for it.

If adding more RAM is not an option, reduce the `--mesh-octree-depth` parameter to go past that step.

journeyman44, 2017-11-19:
Thank you!

I have 16Gb on the host machine, so I could probably move the virtual machine up to 10/12Gb.

I'll give that a go later tonight and update.

Thanks for the support.

journeyman44, 2017-11-20:
@pierotofy a different error this time. I upped the memory by another Gb or so, and dropped octree-depth a little. I have attached the new output.

errors on:

Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

Any new ideas?

[task_output.txt](https://github.com/OpenDroneMap/WebODM/files/1486739/task_output.txt)

pierotofy, 2017-11-20:
Also memory. Resizing the images to a lower resolution would allow you to go past this step if you can't add more memory. We used to have a "resize-to" parameter, but we removed some time ago. We are considering re-adding it for this reason. https://github.com/OpenDroneMap/OpenDroneMap/issues/706

journeyman44, 2017-11-20:
Thanks, Piero.

That seems to have worked.

Time to start playing with parameters to try and get the ortho and the 3d model looking better.

Cheers.

closed 2017-12-22 by pierotofy