WebODM/WebODM #335: Exited with Processing Error 1

mjevanmarle, 2017-11-20:
### How did you install WebODM? (Docker, natively, ...)?
Installed via Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
FireFox 52 on Windows 7

### What is the problem?
The script gets stuck somewhere in the incremental reconstruction: see below

[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm reconstruct /var/www/data/e082c431-abe7-4916-b623-6a1dd966364d/opensfm
2017-11-20 09:28:39,142 Starting incremental reconstruction

2017-11-20 09:28:39,978 Common tracks: 1090
2017-11-20 09:28:40,045 Two-view reconstruction inliers 1082
2017-11-20 09:28:40,162 Triangulated: 1087
2017-11-20 09:28:40,797 Ceres Solver Report: Iterations: 50, Initial cost: 4.668900e+01, Final cost: 3.994001e+01, Termination: NO_CONVERGENCE
2017-11-20 09:28:40,800 Bundle setup/run/teardown 0.0067720413208/0.379173994064/0.00315690040588
2017-11-20 09:28:40,806 -------------------------------------------------------
2017-11-20 09:28:40,816 20171114-104215_DJI_FC300X_DJI_0014.JPG resection inliers: 552 / 557
2017-11-20 09:28:40,830 Adding 20171114-104215_DJI_FC300X_DJI_0014.JPG to the reconstruction
2017-11-20 09:28:41,784 Ceres Solver Report: Iterations: 50, Initial cost: 6.938895e+01, Final cost: 6.296003e+01, Termination: NO_CONVERGENCE
2017-11-20 09:28:41,790 Bundle setup/run/teardown 0.0141401290894/0.862434864044/0.00559091567993
2017-11-20 09:28:41,791 Removed outliers: 0
2017-11-20 09:28:41,821 -------------------------------------------------------
2017-11-20 09:28:41,837 20171114-104217_DJI_FC300X_DJI_0015.JPG resection inliers: 903 / 929
2017-11-20 09:28:41,869 Adding 20171114-104217_DJI_FC300X_DJI_0015.JPG to the reconstruction
2017-11-20 09:28:42,720 Ceres Solver Report: Iterations: 20, Initial cost: 3.555697e+01, Final cost: 2.318229e+01, Termination: CONVERGENCE
2017-11-20 09:28:42,730 Bundle setup/run/teardown 0.0230720043182/0.731673955917/0.00962901115417
2017-11-20 09:28:42,732 Removed outliers: 1
2017-11-20 09:28:42,756 -------------------------------------------------------
2017-11-20 09:28:42,779 20171114-104218_DJI_FC300X_DJI_0016.JPG resection inliers: 1281 / 1300
2017-11-20 09:28:42,817 Adding 20171114-104218_DJI_FC300X_DJI_0016.JPG to the reconstruction

2017-11-20 09:28:45,590 Bundle setup/run/teardown 0.0382380485535/2.6265809536/0.0145370960236
2017-11-20 09:28:45,594 Removed outliers: 6
2017-11-20 09:28:45,623 -------------------------------------------------------
2017-11-20 09:28:45,656 20171114-104221_DJI_FC300X_DJI_0017.JPG resection inliers: 1842 / 1867
2017-11-20 09:28:45,710 Adding 20171114-104221_DJI_FC300X_DJI_0017.JPG to the reconstruction
2017-11-20 09:28:49,347 Ceres Solver Report: Iterations: 50, Initial cost: 2.677067e+02, Final cost: 2.442471e+02, Termination: NO_CONVERGENCE
2017-11-20 09:28:49,363 Bundle setup/run/teardown 0.0520930290222/3.52778816223/0.0163218975067
2017-11-20 09:28:49,368 Removed outliers: 4
2017-11-20 09:28:49,448 -------------------------------------------------------
2017-11-20 09:28:49,456 20171114-104204_DJI_FC300X_DJI_0009.JPG resection inliers: 401 / 409
2017-11-20 09:28:49,472 Adding 20171114-104204_DJI_FC300X_DJI_0009.JPG to the reconstruction

2017-11-20 09:28:52,495 Bundle setup/run/teardown 0.0779390335083/2.79822397232/0.0207421779633
2017-11-20 09:28:52,502 Removed outliers: 2
2017-11-20 09:28:52,541 -------------------------------------------------------
2017-11-20 09:28:52,557 20171114-104202_DJI_FC300X_DJI_0008.JPG resection inliers: 850 / 854
2017-11-20 09:28:52,607 Adding 20171114-104202_DJI_FC300X_DJI_0008.JPG to the reconstruction

2017-11-20 09:28:57,852 Bundle setup/run/teardown 0.0669989585876/5.01042914391/0.0287230014801
2017-11-20 09:28:57,860 Removed outliers: 2
2017-11-20 09:28:57,910 -------------------------------------------------------
2017-11-20 09:28:57,947 20171114-104201_DJI_FC300X_DJI_0007.JPG resection inliers: 1948 / 1955
2017-11-20 09:28:58,006 Adding 20171114-104201_DJI_FC300X_DJI_0007.JPG to the reconstruction

2017-11-20 09:29:03,113 Bundle setup/run/teardown 0.0940809249878/4.88885498047/0.0341410636902
2017-11-20 09:29:03,122 Removed outliers: 1
2017-11-20 09:29:03,190 -------------------------------------------------------
2017-11-20 09:29:03,211 20171114-104156_DJI_FC300X_DJI_0006.JPG resection inliers: 979 / 1091
2017-11-20 09:29:03,252 Adding 20171114-104156_DJI_FC300X_DJI_0006.JPG to the reconstruction

2017-11-20 09:29:06,190 Bundle setup/run/teardown 0.134706020355/2.68358588219/0.0506761074066
2017-11-20 09:29:06,204 Removed outliers: 3
2017-11-20 09:29:06,301 -------------------------------------------------------
2017-11-20 09:29:06,325 20171114-104154_DJI_FC300X_DJI_0005.JPG resection inliers: 1002 / 1014
2017-11-20 09:29:06,367 Adding 20171114-104154_DJI_FC300X_DJI_0005.JPG to the reconstruction

2017-11-20 09:29:10,575 Bundle setup/run/teardown 0.133820056915/3.9007229805/0.0442090034485
[... 80 lines cut ...]
  File "/code/SuperBuild/src/opensfm/opensfm/align.py", line 80, in align_reconstruction_naive_similarity
    T = tf.superimposition_matrix(X.T, Xp.T, scale=True)
  File "/code/SuperBuild/src/opensfm/opensfm/transformations.py", line 1041, in superimposition_matrix
    scale=scale, usesvd=usesvd)
  File "/code/SuperBuild/src/opensfm/opensfm/transformations.py", line 954, in affine_matrix_from_points
    u, s, vh = numpy.linalg.svd(numpy.dot(v1, v0.T))
  File "/usr/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 1359, in svd
    u, s, vt = gufunc(a, signature=signature, extobj=extobj)
  File "/usr/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 99, in _raise_linalgerror_svd_nonconvergence
    raise LinAlgError("SVD did not converge")
numpy.linalg.linalg.LinAlgError: SVD did not converge
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 115, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

### What should be the expected behavior?
No error during process and complete job.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
I uploaded 35 figures made from a drone indoor (camera: dji fc300x | dimensions: 4000 x 3000 | focal: 3.61 | ccd: 6.2 | lat: 0.0 | lon: 0.0 | alt: -95.131) and used default options. I varied with min-num-features up tot 10000 and enabling/disabling use-pmvs option. Besides that I increased the memory of the virtual machine up to 4898 MB, and CPUs with 1 and 2. I also varied with the number of photos, but all have the same error message.Finally I also added GCP info to see if that makes a change, but it didn't.

pierotofy, 2017-11-20:
Hey @mjevanmarle would you be able to share the images with us (upload it on Google Drive)? I've never seen this particular error.

mjevanmarle, 2017-11-21:
Thank you for your response @pierotofy. Overnight I could process a subset of 14 pictures (all with the same angle), however another subset of 14 photos with GCP points returns (again) the same error. I have the feeling that it has to do with the available disk space. Do you know how I can change this? The photos can be found here: https://www.dropbox.com/sh/cqdq8ago9ufgumk/AABkYewnptCjNKqaiEjGQghka?dl=0

pierotofy, 2020-04-02:
Images are no longer available, and ODM has improved a lot since. I will assume this has been fixed. If not, re-open a new issue in the https://github.com/OpenDroneMap/ODM/issues repo. :clinking_glasses:

closed 2020-04-02 by pierotofy