WebODM/WebODM #340: Automated Undistortion and Camera Calibration (and public database)

pierotofy, 2017-11-22:
People are often getting distorted results because they do not undistort camera images before feeding them into WebODM (nor should they have to).

WebODM should offer an intuitive way to:
 - Undistort images based on camera parameters
 - Extract camera parameters based on a chessboard pattern
 - Create a public database of camera parameters and allow users to easily contribute their results

pierotofy, 2017-11-22:
A workflow could be similar to that of Agisoft Lens. http://downloads.agisoft.ru/lens/doc/en/lens.pdf

pierotofy, 2017-11-22:
Note also that opensfm already does camera calibration, and perhaps it's just a matter of tuning the configuration better there.

dakotabenjamin, 2017-11-27:
What can we glean from the lensfun database?

pierotofy, 2017-11-27:
Lensfun is a good starting point, but the process to add new cameras in the database is painful.

juvinski, 2018-03-03:
Hi @pierotofy and @dakotabenjamin  how are this issue ? How can I help ?

pierotofy, 2020-06-03:
Closing this due to infeasibility; the benefits of such system would be minimal and the burden of creating a database would be large.

closed 2020-06-03 by pierotofy

smathermather, 2020-06-03:
Agreed: consensus is that arguably for any commodity cameras, this would provide 0 benefit. For fixed lens, calibrated cameras, the user should be bringing their own. Anecdotally, even a $20k Hasselblad can have variations between cameras in focal distance and principle point of 10%.

pierotofy, 2020-06-03:
Yep.

kf4mot, 2020-06-03:
I'm a very new user to ODM, but I come with a astro-imaging background.
In that world it's normal for a user to shoot calibration images (flats, bias, darks) that are used to take care of things like dust motes, vignetting, distortion . I was a little surprised to see a similar calibration wasn't needed for ODM.

smathermather, 2020-06-04:
It surprised me too and flew in the face of everything I knew about photogrammetry.

The challenge for most drone users (the higher end DJI, some Sensefly and other exceptions perhaps withstanding) is that he cameras are so small and light, so subject to vibration and temperature variation, that it isn't that ODM wouldn't benefit from calibration information, but that one would probably have to calibrate for each flight, or worse. 😄 

So, we recommend instead good flight planning to allow for good self-calibration.