WebODM/WebODM #343: Upload got stuck every time at some point (Showing incredibly high file size)

aigeo, 2017-11-25:
I've installed WebODM on Docker 17 CE installed on Ubuntu 16.04 (DigitalOcean Droplet). SSL Working for my domain. Made Default Port to 443.
### What is the problem?
I was trying to add some photo's to a project but it was not uploading. Upload got stuck at some point. Also the it shows the file size incredibly high. All the photos are optimized and maybe around 800MB.

Please Help
[image]

pierotofy, 2017-11-25:
Hi @aigeo,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your ticket has to contain in order to be processable.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2017-12-02 05:45) I'll close this ticket so it doesn't clutter the bug tracker.

Best regards,
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Neil-99, 2017-11-28:
I was getting the same issue.  I had canceled an upload.  Afterward, everything was claimed to be huge and stalled out.  I rebooted and the problem went away.

pierotofy, 2017-12-03:
Since apparently some of the required information is still missing, I'm closing this now, sorry. Feel free to reopen this or create a new issue once you can provide **all** required information.

closed 2017-12-03 by pierotofy