WebODM/WebODM #344: Stops uploading images

thomas-bamford, 2017-11-25:
### How did you install WebODM? (Docker, natively, ...)?

Docker with SSL on Ubuntu 16.04 LTS Desktop. Updated to latest version. Installation steps taken available upon request.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 57 on Linux, Chrome 62 on Linux

### What is the problem?

Stops uploading images (only reaches 5-10% before stopping) when adding new task when computer is on wifi. The dataset is only 492 Mb in size with 74 images but the upload size reports 34 Gb (See screenshot here [here](https://goo.gl/djHVyL)).

### What should be the expected behavior?

Two days ago (prior to updating to SSL support) the same dataset was able to be uploaded with no issue (when computer is on wifi).

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Default parameters except skipping resize. If resize is selected same result (stops uploading after resize).

thomas-bamford, 2017-11-25:
Sorry forgot to check for similar issues - see #343

pierotofy, 2017-11-25:
Hi @thomas-bamford, thank you for reporting the problem! This is definitely a software fault, I opened a PR that should address this problem #345. I will likely merge this by EOD and rebuild the docker containers.

closed 2017-11-25 by pierotofy

pierotofy, 2017-11-25:
@thomas-bamford could you try to update and confirm that this is now resolved?

thomas-bamford, 2017-11-25:
@pierotofy updated and can confirm that this issue is now resolved.

PR #345 (2017-11-25): Upload timeout fix, file size reporting fix
Fixes upload timeout issue caused by recent Dropzone.js update, as well as the file reporting size.

Should fix #344 #343