WebODM/WebODM #347: Worker exiting (PID:442), Booting worker with PID: 443

aigeo, 2017-11-29:
### How did you install WebODM? (Docker, natively, ...)?

Docker
git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input
cd WebODM
./webodm.sh start

### What's your browser and operating system? 
Chrome Browser.
Ubuntu 16.04
64GB RAM, 6 Core CPU, GTX 1070

### What is the problem?

Task Started but stucked every time.
Saying Autorestarting worker after current request.
Worker exiting (PID: xxx)
Booting worker with PID: xxx
continuous...

### What should be the expected behavior?

Please help

### How can we reproduce this? (What steps did you do to trigger the problem? )
High Quality and DTM checked

pierotofy, 2017-11-29:
Hi @aigeo, "Autorestarting worker after current request." is not an indicator that the task is stuck, it happens on a recurring base to clear out memory from the scheduler.

More importantly, what is the output of the task as seen from the WebODM dashboard?

aigeo, 2017-11-29:
it is there for last 3 hours

2017-11-30 01:26:08,752 Robust matching time : 0.00151586532593s
2017-11-30 01:26:08,753 Full matching 4728 / 4738, time: 3.74363589287s
2017-11-30 01:26:09,305 DJI_0284.JPG - DJI_0287.JPG has 1359 candidate matches
2017-11-30 01:26:09,307 Robust matching time : 0.000760078430176s
2017-11-30 01:26:09,307 Full matching 1307 / 1359, time: 3.54403185844s
2017-11-30 01:26:09,432 DJI_0972.JPG - DJI_0973.JPG has 3220 candidate matches
2017-11-30 01:26:09,433 Robust matching time : 0.00145292282104s
2017-11-30 01:26:09,434 Full matching 3214 / 3220, time: 4.36820602417s

pierotofy, 2017-11-29:
Mm, try restarting WebODM. I'm not entirely sure why this would happen, does it happen consistently or just with certain datasets?

`./webodm.sh down && ./webodm.sh start`

aigeo, 2017-11-30:
It was happening consistently, Here is the data set
[https://drive.google.com/open?id=1NV-Ei459hBV_dK_1xLCz2xI0J3h_P-AY](url)

I left my computer turned on and in the morning i found it was not working, i mean hanged. Then I restarted and installed again on windows OS. Same issue there also.

Trying with a new dataset again

aigeo, 2017-11-30:
With another dataset

2017-11-30 18:34:09,188 Bundle setup/run/teardown 1.36422801018/10.7334640026/0.353708028793
2017-11-30 18:34:09,297 Removed outliers: 3
2017-11-30 18:34:10,076 -------------------------------------------------------
2017-11-30 18:34:10,117 DJI_0649.JPG resection inliers: 3096 / 3100
2017-11-30 18:34:10,146 Adding DJI_0649.JPG to the reconstruction

Stucked here for 2.5 hours 

Please help

pierotofy, 2017-12-02:
I wish I was able to reproduce this. I would try to decrease the number of images, or resize them to a smaller resolution and see if it's a problem related to the dataset size. I really don't know why it's stuck.

From the console (the window where you type `./webodm.sh start`) is there any error messages?

aigeo, 2017-12-03:
no, the starting window was okay without error. I've reinstalled multiple times and tried, i'm not sure WebODM docker can utilize the full power of my 12 core and 64 GB RAM.

Another task i started friday morning and now it stucked here,

[image]

rulabmarton, 2018-03-24:
Hy everybody, a faced the same issue. I don't know why, but if start webodm without --no-debug parameter it works.

EpicLPer, 2019-04-26:
Not sure if this is related but mine got stuck at:
```
2019-04-26 20:38:51,696 INFO: DJI_0189.JPG resection inliers: 2228 / 2256
2019-04-26 20:38:51,760 DEBUG: Ceres Solver Report: Iterations: 3, Initial cost: 1.554498e+02, Final cost: 1.272979e+02, Termination: CONVERGENCE
2019-04-26 20:38:51,760 INFO: Adding DJI_0189.JPG to the reconstruction
2019-04-26 20:38:52,006 INFO: Re-triangulating
```

pierotofy, 2020-04-02:
WebODM now employs redis locks, which should prevent these kind of issues. If the issue is still there, please re-open. :clinking_glasses:

closed 2020-04-02 by pierotofy