WebODM/WebODM #355: Process completed but isn't good.

cfdar, 2017-12-02:
Hi, can anyone help me. I made a process with this configuration
"Options: mesh-octree-depth: 12, min-num-features: 8000, orthophoto-resolution: 60, dem-resolution: 0.04, dsm: true, use-25dmesh: true, use-pmvs: true"
The result wasnt good, the roofs are curv. 
In the link to google drive have the individual images, the ortomosaic results from WebODM and other i procesed with dronedeploy. [https://drive.google.com/drive/folders/1A6yRorRe1uReqf6NmnYhRqxsD2h2imxF?usp=sharing](url)

pierotofy, 2017-12-02:
Hi @cfdar,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your ticket has to contain in order to be processable.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2017-12-09 08:45) I'll close this ticket so it doesn't clutter the bug tracker.

Best regards,
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2017-12-02:
Hey @cfdar, try to remove the use-25dmesh option from your preset.

cfdar, 2017-12-04:
Hi @pierotofy , i removed the use-25dmesh option, and the result was better, but isn't good like drone deploy, the roofs are stiil curved, less than before, and the vegetation looks like blurred. Do you have a default parameters for images from urban area ou vegetation area?

Where can i find a manual with details about task option from WebODM?

pierotofy, 2017-12-04:
Here's a Wiki page on the task options: https://github.com/OpenDroneMap/OpenDroneMap/wiki/Run-Time-Parameters

We are constantly working to improve the quality of the outputs, so keep an eye on the latest developments.

closed 2017-12-04 by pierotofy