WebODM/WebODM #358: About WebODM parameters

jesserovira, 2017-12-04:
Hi,

Could you, please, show us the parameters and values that you use at Lab, when you are going to test and release?

The product is great and your iniative speakless. I thanks a lot your contribution until now. The only thing is that I'm not achiving a result with a good quality, and I'm pretty sure it's because the parameters. If you could reveal what are you using on yours tests, could improve my tests and decision step to go definitely to WebODM platform.

Until now, our tests were running at Docker, last release, under UBUNTU 16.04, with a test environment, with Django also.

Thank you!

pierotofy, 2017-12-04:
Hi @jesserovira,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your ticket has to contain in order to be processable.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2017-12-11 15:15) I'll close this ticket so it doesn't clutter the bug tracker.

Best regards,
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2017-12-05:
Hi @jesserovira, we currently have only this document https://github.com/OpenDroneMap/OpenDroneMap/wiki/Run-Time-Parameters. Since the codebase is changing frequently at the moment, we haven't had time to document it properly.

I agree that we should have a document that explains the parameters in more detail (and with less technical information).

Tuning the parameters is more art than science (it's trial and error, knowing the dataset and understanding the effects of each parameter). I personally start with WebODM's Default preset and make adjustments from there.

pierotofy, 2020-04-02:
Plenty of resources are now available that cover WebODM parameters: https://www.opendronemap.org/docs/, https://odmbook.com

closed 2020-04-02 by pierotofy