WebODM/WebODM #359: Uploading photos from a smartphone

Marcuzzo77, 2017-12-05:
Hello to all,

I have problems uploading photos from smartphones.
This happens on Android and Apple devices.
I have tried different browsers but it does not change anything.
Help?

pierotofy, 2017-12-05:
Hi @Marcuzzo77,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your ticket has to contain in order to be processable.
Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.
I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2017-12-08 14:55) I'll close this ticket so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2017-12-09:
Since apparently some of the required information is still missing, I'm closing this now, sorry. Feel free to reopen this or create a new issue once you can provide **all** required information.

closed 2017-12-09 by pierotofy

Marcuzzo77, 2017-12-10:
How did you install WebODM? (Docker, natively, ...)?

Natively

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Safari on Iphone 6s

What is the problem?

I have problems uploading photos from smartphones.
This happens on Android and Apple devices.
I have tried different browsers but it does not change anything.
Help?

What should be the expected behavior?

...

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

...