WebODM/WebODM #360: Could not find ccd_width in file with MAPIR SURVEY2 NDVI camera

DroneMapper, 2017-12-06:
### How did you install WebODM? (Docker, natively, ...)?

[With Webodm installer]

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Google Chrome]

### What is the problem?

 [Could not find ccd_width in file with MAPIR survey 2 
[image]
[image]
[image]

camera]

[INFO]    Initializing OpenDroneMap app - Wed Dec 06 16:41:23  2017
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/7fce9c33-8f8b-4855-8936-bf10a14f2143/images
[DEBUG]   /var/www/data/7fce9c33-8f8b-4855-8936-bf10a14f2143/images
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174348_014.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8417365 | lon: 25.9687783 | alt: 1376.9
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174404_022.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8422151 | lon: 25.9701127 | alt: 1377.33
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174338_008.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8414666 | lon: 25.9680251 | alt: 1375.52
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174332_004.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8412581 | lon: 25.9674381 | alt: 1374.52
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174327_002.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8411102 | lon: 25.9670277 | alt: 1373.97
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174407_024.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8423345 | lon: 25.9704465 | alt: 1376.99
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174354_018.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8419468 | lon: 25.9693636 | alt: 1378.02
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174357_020.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8420374 | lon: 25.9696137 | alt: 1377.77
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174345_012.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8416469 | lon: 25.968527 | alt: 1376.08
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174351_016.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8418557 | lon: 25.9691142 | alt: 1377.85
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174341_010.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8415567 | lon: 25.9682757 | alt: 1375.95
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_1105_174335_006.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 4.0 | ccd: None | lat: -27.8413779 | lon: 25.9677735 | alt: 1375.06
[Type answer here]

### What should be the expected behavior?

[Type answer here]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[Using the default option]

dakotabenjamin, 2017-12-06:
You can find the Sensor Width [here](https://www.mapir.camera/pages/cameras) for your camera, then plug that into the "force-ccd" option

DroneMapper, 2017-12-06:
Thanks that sorted out the ccd error, 

The dimentions of the camera is 4608x3456 but when processing is dimensions: 2048 x 1536

[DEBUG]   Loaded 2017_1105_174348_014.JPG | camera: mapir camera survey2_ndvi | dimensions: 204
imensions: 2048 x 1536 | focal: 3.97 | ccd: 6.17472 | lat: -27.8417365 | lon: 25.9687783 | alt: 1376.9
[DEBUG]   Loaded 2017_1105_174404_022.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 3.97 | ccd: 6.17472 | lat: -27.8422151 | lon: 25.9701127 | alt: 1377.33
[DEBUG]   Loaded 2017_1105_174338_008.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 3.97 | ccd: 6.17472 | lat: -27.8414666 | lon: 25.9680251 | alt: 1375.52
[DEBUG]   Loaded 2017_1105_174332_004.JPG | camera: mapir camera survey2_ndvi | dimensions: 2048 x 1536 | focal: 3.97 | ccd: 6.17472 | lat: -27.8412581 | lon: 25.9674381 | alt: 1374.52

dakotabenjamin, 2017-12-07:
Did you resize the images when you uploaded them?

pierotofy, 2017-12-10:
Closing this for now, please use the forum if you have further questions.

closed 2017-12-10 by pierotofy