WebODM/WebODM #361: Uploading photos from a smartphone

Marcuzzo77, 2017-12-10:
How did you install WebODM? (Docker, natively, ...)?

Natively

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Safari 11 on iOS11

What is the problem?

I have problems uploading photos from smartphones.
This happens on Android and Apple devices.
I have tried different browsers but it does not change anything.
Help?

What should be the expected behavior?

...

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I try to upload some photos from my smartphone remotely having natively installed on the server the application from which I can access in any way but from any smartphone and browser can not upload neache a photo

pierotofy, 2017-12-10:
Hey @Marcuzzo77, thanks for reporting the problem. I can confirm this happens on Safari 11 on iOS. 

I was able to upload images using Android 7 while using Chrome however, so it doesn't affect all mobile devices and platforms.

Marcuzzo77, 2017-12-10:
Hi Piero,

in fact, as you say. Tested on other Android devices and works properly. Sin...
I also tried Chrome from iOS but nothing ....

pierotofy, 2017-12-21:
Fixed per #366

closed 2017-12-21 by pierotofy