WebODM/WebODM #362: Exception: Child returned 1

Marcuzzo77, 2017-12-10:
### How did you install WebODM? (Docker, natively, ...)?

Natively

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 62 on Mac OS X 10

### What is the problem?

A dataset from 169 nadiral photos is not processed correctly

### What should be the expected behavior?

...

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

### Report:
[report.txt](https://github.com/OpenDroneMap/WebODM/files/1546037/report.txt)

### Options:
mesh-solver-divide: 15, mesh-octree-depth: 12, min-num-features: 8000, force-focal: 8, resize-to: -1, orthophoto-resolution: 20, pmvs-csize: 8, mesh-samples: 5, use-25dmesh: true, opensfm-processes: 12, use-pmvs: true

### Dataset:
https://drive.google.com/drive/folders/12fW1FX2sZUF5F1Kv6anGAFjcWJ64UsaE

pierotofy, 2017-12-17:
Hey @Marcuzzo77, thanks for reporting the problem.

I'll have to look further into this, since I've never seen this error (`numpy.linalg.linalg.LinAlgError: SVD did not converge`) before.

Marcuzzo77, 2017-12-17:
Hi Piero,

I am pleased to somehow contribute to the improvement of the code. I’m looking forward to hearing from you

pierotofy, 2017-12-21:
I had a chance to look at the data (hey, so close to home, I'm originally from Trieste) 😄 I'm willing to bet the problem is due to the fact that many images share the lat/lon data in the EXIF parameters. If you want to use ground control points, you should provide a `gcp_list.txt` file as indicated in https://github.com/OpenDroneMap/OpenDroneMap/wiki/Running-OpenDroneMap#running-odm-with-ground-control.

Marcuzzo77, 2017-12-21:
Hello Piero,

I imagined that you were from my areas by your last name. As for the processing of the cloud , I will try to insert the GCP. My friends and me are trying to make a system that could handle them automatically.
P.S.I live in Udine. Let me know when you are near this area, I’d like to met you.

pierotofy, 2017-12-21:
For sure!

Take a look at https://github.com/wolkstein/OpenDroneMap-GCP_LIST.TXT-generator, some work has already been started on automatic GCP detection.

closed 2018-01-15 by pierotofy