WebODM/WebODM #363: Task killed at end of processing

CrileNZL, 2017-12-11:
### How did you install WebODM? (Docker, natively, ...)?

Run via Docker on a 64-bit Win7 laptop with 8 Gb RAM.  I've allocated 5 Gb of RAM to the VM and resized storage to 100 Gb.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 56 on Windows 7

### What is the problem?

After two hours processing 40 images, task was killed.  A subset of the first five images in the dataset ran to completion though results weren't great (orthophoto was very patchy and data layers were offset from true location)

### What should be the expected behavior?

Task runs to completion

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

dataset [here](https://drive.google.com/drive/folders/1sbncW_TZ5mJ9jBiGhfUv4RpS30XOs6az?usp=sharing)

Parameters were set to:

- min-num-features: 100000
- resize-to: 2048
- dem-terrain-type: FlatNonForest
- dsm: true
- opensfm-processes: 1
- use-pmvs: true

It seemed like it was getting to the end of the processing - console output showed multiple variations of:  "webapp    | WARNING Run time of job "background.<locals>.wrapper <trigger: interval[0:00:05], next run at: 2017-12-11 05:52:17 UTC>" was missed by 0:00:03.27742".

task_output.txt attached.
[task_output.txt](https://github.com/OpenDroneMap/WebODM/files/1546770/task_output.txt)

I've had a difficult time getting WebODM configured properly and hadn't seen "Killed" in any outputs previously.  Any suggestions are welcome.

Crile

pierotofy, 2017-12-11:
Hi @CrileNZL, WebODM should have showed a message like this:

```
It looks like your processing node ran out of memory. If you are using docker, make sure that your docker environment has enough RAM allocated.```

And the suggestion is accurate, you need more memory to process that dataset at the resolution you are trying to process it. Note that something is off because you say you passed the `resize-to` parameter, but your images are being processed at 6000 x 4000. Make sure you've updated your node-opendronemap and WebODM docker images to the latest version.

`./webodm.sh update`

CrileNZL, 2017-12-12:
Thanks Peiro,

I’m struggling to see how I can increase my RAM – the virtual box has been allocated 5 of my 8 Gb and I have ensured the resize is at 2048 x 1365 as well as resetting the orthophoto-resolution from 20 to 5.  If I understand that parameter correctly, it’s the number of pixels per metre so 5 should be a lower resolution than the default, correct?  I’ve also reduced the number of images from 30 to 40.  Before this set of runs I ran ./webodm.sh update so I should be up to date with the latest versions.

Still crashing out of memory.

Thanks –

C

P Please consider the environment before you print this email.
"The contents of this e-mail (including any attachments) may be confidential and/or subject to copyright. Any unauthorised use, distribution, or copying of the contents is expressly prohibited. If you have received this e-mail in error, please advise the sender by return e-mail or telephone and then delete this e-mail together with all attachments from your system."

pierotofy, 2017-12-12:
5GB about the max you'll be able to assign to a 8GB host. Also reduce `min-num-features` (leave the default, unless you're not getting good results), 100000 is probably way too high.

CrileNZL, 2017-12-12:
A quick follow up. I’ve rerun WebODM with the following parameters as suggested:

·         Min-num-features: 6000

·         Orthophoto-resolution: 5

·         Dem-resolution: 1

·         Dem-terrian-type: FlatNonForest

·         Use-pmvs: true

Task_output.txt attached.

While running this I had a resource monitor open and while it ran (for 00:01:48) I had between 3 and 3.5 Gb free in working memory.  Is there somewhere else I could be getting tight on memory?

Thanks –

C

[INFO]    Initializing OpenDroneMap app - Tue Dec 12 21:21:40  2017
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/d9365028-7c25-47a3-939f-02524734c118/images
[DEBUG]   /var/www/data/d9365028-7c25-47a3-939f-02524734c118/images
[DEBUG]   Loaded DSC00184.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6467764999 | lon: 172.462495 | alt: 69.9
[DEBUG]   Loaded DSC00166.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6473281 | lon: 172.4624356 | alt: 69.5
[DEBUG]   Loaded DSC00169.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.647026 | lon: 172.4630687 | alt: 69.7
[DEBUG]   Loaded DSC00175.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6467073 | lon: 172.4631862 | alt: 70.2
[DEBUG]   Loaded DSC00165.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6473396001 | lon: 172.4629684 | alt: 70.4
[DEBUG]   Loaded DSC00164.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6472356999 | lon: 172.4631784 | alt: 70.7
[DEBUG]   Loaded DSC00163.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6471311999 | lon: 172.4633885 | alt: 70.8
[DEBUG]   Loaded DSC00176.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6468109 | lon: 172.4629762 | alt: 70.2
[DEBUG]   Loaded DSC00187.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6464689 | lon: 172.4631171 | alt: 70.3
[DEBUG]   Loaded DSC00171.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6468162 | lon: 172.4635001 | alt: 69.2
[DEBUG]   Loaded DSC00179.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6471241 | lon: 172.4623472 | alt: 70.2
[DEBUG]   Loaded DSC00189.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.646203 | lon: 172.4631351 | alt: 70.3
[DEBUG]   Loaded DSC00180.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6472232999 | lon: 172.4621448 | alt: 69.7
[DEBUG]   Loaded DSC00173.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6465046 | lon: 172.4636032 | alt: 69.7
[DEBUG]   Loaded DSC00190.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6463056 | lon: 172.4629154 | alt: 70.6
[DEBUG]   Loaded DSC00177.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6469154001 | lon: 172.4627647 | alt: 70.3
[DEBUG]   Loaded DSC00191.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6464066 | lon: 172.462714 | alt: 70.5
[DEBUG]   Loaded DSC00185.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6466754 | lon: 172.4627016 | alt: 69.9
[DEBUG]   Loaded DSC00167.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6472286 | lon: 172.4626458 | alt: 70.4
[DEBUG]   Loaded DSC00181.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6470842 | lon: 172.4618722 | alt: 69.7
[DEBUG]   Loaded DSC00178.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6470186999 | lon: 172.4625547 | alt: 70.4
[DEBUG]   Loaded DSC00188.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.646363 | lon: 172.463321 | alt: 70.6
[DEBUG]   Loaded DSC00170.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6469198999 | lon: 172.4632886 | alt: 69.4
[DEBUG]   Loaded DSC00174.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6466048999 | lon: 172.4633914 | alt: 70.1
[DEBUG]   Loaded DSC00183.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6468781999 | lon: 172.4622925 | alt: 70.0
[DEBUG]   Loaded DSC00162.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6470274 | lon: 172.4635929 | alt: 70.7
[DEBUG]   Loaded DSC00168.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6471268999 | lon: 172.4628624 | alt: 70.2
[DEBUG]   Loaded DSC00182.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6469873001 | lon: 172.4620703 | alt: 70.0
[DEBUG]   Loaded DSC00172.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6467177 | lon: 172.4637019 | alt: 69.3
[DEBUG]   Loaded DSC00186.JPG | camera: sony ilce-5100 | dimensions: 2048 x 1365 | focal: 0.0 | ccd: 23.5 | lat: -43.6465717999 | lon: 172.4629098 | alt: 70.0
[INFO]    Found 30 usable images
[... cut ...]
    fid = open(file, "rb")
IOError: [Errno 2] No such file or directory: '/var/www/data/d9365028-7c25-47a3-939f-02524734c118/opensfm/features/DSC00184.JPG.npz'
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 99, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

pierotofy, 2018-03-02:
If there are further questions about this subject,  according to the [guidelines](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md) please let's move them to the forum. http://community.opendronemap.org/c/webodm

closed 2018-03-02 by pierotofy