WebODM/WebODM #364: Unable to make migrations for new model

Pratyush1991, 2017-12-17:
### How did you install WebODM? (Docker, natively, ...)?
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 57 on windows 10

### What is the problem?

Unable to migrate new model in development environment

### What should be the expected behavior?

Need help in migrating models 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I am currently running webodm in development environment. I added a new model in app but I am unable to migrate the models. Screenshot attached below.

[image]

[image]

I did run python manage.py makemigrations.
[image]

pierotofy, 2017-12-17:
You are running the makemigrations command from your host environment, which is not going to work (you need to run the command within the webodm_webapp docker container).

Or more simply, just recreate the container:

`./webodm.sh down && ./devenv.sh start`

Pratyush1991, 2017-12-18:
Thanks a lot. I did try to recreate the container using above command still the changes were not reflected in the database not sure why. So I ran the migration command inside the docker container which solved my problem.

I would also like to know if I should post issues here or somewhere else for better communication as I am a bit new to development environment and often have some questions or queries.

pierotofy, 2017-12-18:
If there's a bug, this is the best place to report it. If it's more of a question, the forum at http://community.opendronemap.org/c/webodm is probably the better place. 👍

closed 2017-12-18 by pierotofy