WebODM/WebODM #365: Can't upload images

tareknitro, 2017-12-20:
Hello , i have a problem with WebODM ,i can't upload  the images ,I do not see the loading bar and i can't start processing , i use ubuntu trusty , some one can help ?

pierotofy, 2017-12-20:
Hi @tareknitro,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2017-12-23 12:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

tareknitro, 2017-12-20:
/home/yovan/Bureau/Capture du 2017-12-20 18:19:09.png
/home/yovan/Bureau/Capture du 2017-12-20 18:28:16.png
i use webodm with docker 
Mozila browser

pierotofy, 2017-12-24:
Since apparently some of the required information is still missing, I'm closing this now, sorry. Feel free to reopen this or create a new issue once you can provide **all** required information.

closed 2017-12-24 by pierotofy