WebODM/WebODM #370: Error with ./webodm.sh start

MarqFerrJr, 2018-01-09:
Hey there. I've installed WebODM with docker on Ubuntu 16.04.

When I put the "./webodm.sh start" command line, I have the following:

Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Starting db         ... 
Starting node-odm-1 ... 
Starting webapp     ... error

ERROR: for db  ('Connection aborted.', error(13, 'Permission denied'))

ERROR: for node-odm-1  ('Connection aborted.', error(13, 'Permission denied'))
ERROR: Couldn't connect to Docker daemon - you might need to run `docker-machine start default`.
ERROR: Couldn't connect to Docker daemon - you might need to run `docker-machine start default`.

I tried to run "docker-machine start default command", but the output always says it' already running, I'm stuck with this, I really don't know what to do anymore

Please, Somebody could help me with this issue?

pierotofy, 2018-01-09:
Hi @MarqFerrJr,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-01-12 12:40) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2018-01-13:
Since apparently some of the required information is still missing, I'm closing this now, sorry. Feel free to reopen this or create a new issue once you can provide **all** required information.

closed 2018-01-13 by pierotofy