WebODM/WebODM #371: Error: No machine name(s) specified and no "default" machine exists

renandp11, 2018-01-11:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 63 on Windows 10
IS YOUR WEB BROWSER UP TO DATE?
✓ Your web browser is up to date.

### What is the problem?

Getting "Error: No machine name(s) specified and no "default" machine exists" after type docker-machine ip  based on (https://github.com/OpenDroneMap/WebODM)

### What should be the expected behavior?

To get the VM IP

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Following https://github.com/OpenDroneMap/WebODM    steps

pierotofy, 2018-01-11:
Hey :hand: @renandp11 if you are using Docker with Hyper-V you don't need to find the IP of your docker-machine (because you're not using docker-machine), just connect to http://localhost:8000 after launching WebODM. Docker is a bit confusing for Windows users, that's why we released an installer at https://www.webodm.org/installer

closed 2018-01-11 by pierotofy

renandp11, 2018-02-09:
> Hey ✋ @renandp11 <https://github.com/renandp11> if you are using Docker

Hi Piero!

Thanks, just a little troubleshooting and got WebODM up and running ... so
cool 😎