WebODM/WebODM #375: No usable processing nodes.

rdmhora, 2018-01-17:
### How did you install WebODM? (Docker, natively, ...)?

Docker
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Safari 11 on Mac OS X 10
### What is the problem?

Hello,

I've been using WebODM in docker for Mac without any problem until I upgraded Docker. I tried to re-install docker and WebODM but didn't Work.
Can anyone help me please.

[image]

[image]

[image]

pierotofy, 2018-01-21:
Does the problem persist if you restart WebODM / docker? Does it happen sporadically or all the times?

rdmhora, 2018-01-21:
Well, It was happing all the time when I tried to run WebODM. I tried to re-install Docker and WebODM and nothing changed. I was using docker's stable version. I uninstalled it and decide to install the edge version and this solved the problem.

tannerpace, 2020-01-28:
what is the edge version? you just changed your browser to microsoft edge? I am having the same problem, I can however process with 2d maps with node micmac. I am would like to be able to make 3d models.

rdmhora, 2020-01-28:
> The edge version? Did you just change your browser to Microsoft's edge? I am having the same issue. I can, however, process with 2d maps with node Micmac. I am would like to be able to make 3d models.

Well, I use a MacBook Pro to run WebODM. I decide to re-install every thing and I start to use Opera Browser. Try to update WebODM and certify that docker is using enough memory. To be honest I don’t remember What I did to solve the problem, but I’m sure that I re-install every software...sorry and I hope my information can help you

lordzyx, 2020-07-25:
./wbodm.sh rebuild worked for me

pierotofy, 2020-12-07:
This has been a while; we've made lots of improvements since 2018. If this is still an issue, please re-open with an updated report? :pray:

closed 2020-12-07 by pierotofy