WebODM/WebODM #376: node-odm-1:3000 seems to be offline.

perro001, 2018-01-21:
[image]

### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 63 on Windows 10

### What is the problem?
Hello,
I Can process images because the node-odm-1:3000 seems to be offline. 
Any ideas on how to solve the problem?
Thank you  

[image]

Thank you for your  help

pierotofy, 2018-01-21:
This looks similar to #375. Usually restarting WebODM / docker will solve the issue, but I wonder if there's something we can do about it to prevent it.

Does this happen often, or just sporadically?

DelphiR, 2018-05-14:
I want to node-odm-1:3000 to be temporarily offline. Could you answer waat was the problem?

pierotofy, 2020-04-02:
Closing due to inactivity. Let's move future conversations over to https://community.opendronemap.org if this is still an issue :+1:

closed 2020-04-02 by pierotofy