WebODM/WebODM #377: Erro between Node and WebODM - zipfile.BadZipFile: File is not a zip file

juvinski, 2018-01-21:
### How did you install WebODM? (Docker, natively, ...)?

Natively

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

https://www.whatismybrowser.com/w/T7TPD35

### What is the problem?

Processing a medium dataset after node finish the zip file transfer between node-odm and webodm is finished but when Webodm try to decompress the file there is an exception:
INFO Done downloading all.zip for Task [Task of 2018-01-21T15:51:54.820Z] (34684e04-066e-4b11-bad4-bdfcc3e8a65d)
ERROR Uncaught error! This is potentially bad. Please report it to http://github.com/OpenDroneMap/WebODM/issues: File is not a zip file Traceback (most recent call last):
  File "/home/agroraptor/RaptorWeb/app/scheduler.py", line 53, in process
    task.process()
  File "/home/agroraptor/RaptorWeb/app/models/task.py", line 383, in process
    with zipfile.ZipFile(zip_path, "r") as zip_h:
  File "/usr/lib/python3.5/zipfile.py", line 1026, in __init__
    self._RealGetContents()
  File "/usr/lib/python3.5/zipfile.py", line 1093, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file

If I manually copy the all.zip file from node and decompress to another folder the all.zip is perfect, just the copy to webodm is wrong.

### What should be the expected behavior?

on webodm the process is complete with success but there is no assets. 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

this is the dataset
https://drive.google.com/open?id=0B1omx8hPUswTOW8tOWdjZFprMG8

juvinski, 2018-01-21:
Hi,

I list compare the files and the zip inside node is greater than webodm
./Node/app/media/project/1/task/34684e04-066e-4b11-bad4-bdfcc3e8a65d/assets# ls -alt
total 2475240
-rw-rw-r-- 1 webodm webodm 2534616101 Jan 21 18:13 all.zip
drwxrwxr-x 2 webodm webodm       4096 Jan 21 18:13 .
drwxrwxr-x 3 webodm webodm      20480 Jan 21 18:13 ..

./Web/app/media/project/1/task/34684e04-066e-4b11-bad4-bdfcc3e8a65d/assets# ls -alt 
-rw-rw-r--  1 webodm webodm 2535477953 Jan 21 18:13 all.zip

pierotofy, 2018-01-22:
Hey @juvinski thanks for reporting the problem. The filesize looks pretty big, does this happen only with large datasets?

juvinski, 2018-01-22:
Hi @pierotofy 
On previous version I didn't have this problem with result images with more than 6Gb.
I'm testing again with a new dataset and I will update.

pierotofy, 2018-01-22:
Mm, it's odd because nothing has changed in the zipping/extraction code for months.

dakotabenjamin, 2018-01-22:
A user on gitter just reported a similar problem using the `zoo` dataset.

juvinski, 2018-01-22:
I have on same machine a dataset of 281 images and is working fine, just this of 321 is throwing this exception.
Maybe a timeout in operation ?

pierotofy, 2018-01-22:
If you restart the task does it throw the error again?

juvinski, 2018-02-05:
Hi @pierotofy , sometime work sometime not. I'm using a google cloud machine, on non VM environment was everthing working pretty fine.

juvinski, 2018-02-07:
Hi @pierotofy , I tried several times and changing some basic parameters - resolution, debug level, but still with this fault copying the zip file.

juvinski, 2019-02-18:
Hi, after several tests, and a birthday cake, I think this was a problem on my structure or something has fixed the issue. I will be closing this one.

closed 2019-02-18 by juvinski