WebODM/WebODM #378: Process Exited with code 132

neosaadmin, 2018-01-23:
### How did you install WebODM? (Docker, natively, ...)?

Docker: 
git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input
cd WebODM
./webodm.sh start

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 63 64-bit

### What is the problem?

I get this code on the dashboard after attempting to upload an image set:

Process exited with code 132

/code/run.sh: line 6:    21 Illegal instruction     (core dumped) python $RUNPATH/run.py "$@"

Everything starts fine, no errors or warnings in either install or running the webodm.sh start. Only issue is after the upload has finished. I have had multiple attempts at the install with no success. I understand that it might need to be built from the source, but I am unable to get a working native install working after multiple attempts. I am attempting to install on a Ubuntu Server 16.04 64 bit server. 48 Core Opteron 6172s 64GB. SOOOO the MMX, SSE, SSE2, etc. Instructions are there. 

I installed in the same manor onto a 64bit Ubuntu server VM on my laptop and it worked without errors. 

Any information on how to repair this would be great, if you have any advice or information in regards to a native install with build instructions, that would also be grand.

Thanks, Great program regardless.

pierotofy, 2018-01-25:
Hey @neosaadmin thanks for reporting the problem.

Effectively it seems like your CPU should meet the requirements, but for some reason there's an instruction that's tripping the program. 

I would follow the instructions on https://github.com/OpenDroneMap/OpenDroneMap/ to compile it ODM from sources, then do the same for node-OpenDroneMap (https://github.com/OpenDroneMap/node-OpenDroneMap). If you get stuck I would recommend checking the issue list on both projects and eventually open a discussion on http://community.opendronemap.org if you need help.

The prebuilt docker images are not going to work.

pierotofy, 2018-01-25:
I'll leave this issue open since we should investigate why certain CPUs that meet gcc's requirements are not working (-march core2): https://gcc.gnu.org/onlinedocs/gcc-4.5.3/gcc/i386-and-x86_002d64-Options.html

DelphiR, 2018-05-10:
Hi. I have made a decade of tests using different computers in our lab.
This error will 100% reproduced in many AMD CPUs from athlon to phenom ones.
The attempt of recompile will not help. It seems that docker contains some Python compiled code for special cpus. Dear autors! You just enable the core2 optimization which may give you 1-2% of calculation speed but destroy compatibility with many CPUs! This is bug and this is 100% incorrect because involved a lot of users!
I also try to rebuild Python packegs with no succes.
It seems that docker was build also with a newer cpu to get such type of errors

pierotofy, 2018-05-10:
@DelphiR there's no need to repost this multiple times.

The `core2` flag is not an optimization, rather it's a de-optimization to make sure that older computers can run the docker images that we compile from newer hardware. This is not a bug, rather a tradeoff decision between speed and supporting hardware.

https://gcc.gnu.org/onlinedocs/gcc-4.5.3/gcc/i386-and-x86_002d64-Options.html

If you need to support an older / unsupported architecture, there's always the option to compile from sources your own docker images. It's not difficult, just follow the README at https://github.com/OpenDroneMap/OpenDroneMap

juvinski, 2018-05-10:
HI, this is happening with mvs-texturing module phase?

DelphiR, 2018-05-11:
Hi. It seems to be not because in my case this happens at the beginning and it is related ti python seems to be. I use ubuntu 14.04 with docker image and athlon X2 cpu for testing. All other equipment with newer cpus works. I also tested the native build image in different cpus and I see this type of bugs in Xeon cpu when I tested it on Azure service.

DelphiR, 2018-05-11:
I try to recompile webodm-node inside the docker without changing anything. And this issue didn't disappeared.

DelphiR, 2018-05-11:
The googling leading me to SSSE3 problem

DelphiR, 2018-05-11:
Hmm. The core2 enables the SSSE3 optimization and SSS3 is lacked in many strong CPU up to 6 cores phenom X6 for example. This table of support SSSE3 which doesn't contains a lot of old popular models.
Xeon 5100 Series
Xeon 5300 Series
Xeon 3000 Series
Xeon E3
Xeon E5
Xeon E7
Core 2 Duo
Core 2 Quad
Core 2 Extreme
Core i3
Core i5
Core i7
Pentium Dual-Core
Celeron (Celeron 4xx, ядро Conroe-L)
Celeron Dual-Core
Celeron M 500 Series
Intel Atom

DelphiR, 2018-05-11:
This is info from manual of march from gcc 4.5.2
core2
Intel Core2 CPU with 64-bit extensions, MMX, SSE, SSE2, SSE3 and SSSE3 instruction set support. 
atom
Intel Atom CPU with 64-bit extensions, MMX, SSE, SSE2, SSE3 and SSSE3 instruction set support

DelphiR, 2018-05-11:
/code/run.sh: line 6:    22 Illegal instruction     (core dumped) python $RUNPATH/run.py "$@"

pierotofy, 2018-08-25:
Possibly fixed by https://github.com/OpenDroneMap/OpenDroneMap/issues/873.

nchamo, 2019-05-23:
Hey @pierotofy !

Just wanted to confirm that I had the same problem, but with the change in OpenDroneMap/ODM#873, it went away. As I explained in that issue, it is still possible to make the docker images work in hardware that doesn't support SSE* instructions, but it needs a few extra steps.

Thanks for the great work!