WebODM/WebODM #381: ./webodm.sh start No such file or directory ?????

BryanNav, 2018-01-28:
### How did you install WebODM? (Docker, natively, ...)?

Ive been following the install instructions from https://github.com/OpenDroneMap/WebODM 

I am on Windows 7 64bit and have gotten as far as the downloading DockerToolbox, Python, and Git all the latest versions. 

On Docker Quickstart I run
 git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input
cd WebODM
./webodm.sh start

The result is ./webodm.sh: line 4: cd: /c/Users/Bryan: No such file or directory

 For this error I found the troubleshooting on the same page: 

Cannot start WebODM via ./webodm.sh start, error messages are different at each retry | You could be running out of memory. Make sure you have enough RAM available. 2GB should be the recommended minimum, unless you know what you are doing

On the VM VirtualBox manager I set the System>Base memory to 4GB and the Storage>Virtual Size to 97.66GB 

I try to run ./webodm.sh start and it gives me the same "no such file or directory available"

I have no experience with command line and I'm sure there is an easy fix to get past this first step but I havent found any additional solutions. Please help Thanks
-- | --

pierotofy, 2018-01-28:
Hi @BryanNav,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-01-31 14:05) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2018-02-01:
Since apparently some of the required information is still missing, I'm closing this now, sorry. Feel free to reopen this or create a new issue once you can provide **all** required information.

closed 2018-02-01 by pierotofy