WebODM/WebODM #383: Ability to stream/access lower resolution models

pierotofy, 2018-01-30:
Necessary for #316 https://github.com/pierotofy/webodm-ar/issues/2

Could also be nice to speed up WebODM's built-in 3D viewer.

pierotofy, 2018-05-16:
Thanks @smathermather! I'm having a hard time finding the multi-resolution extension you mentioned. This is the closest I found in their docs: http://vtsdocs.melown.com/en/latest/reference/server/cmdline-tools/encoders.html

Maybe it's not documented?

smathermather, 2018-05-16:
They should be here at FOSS4G. Let’s ask!

pierotofy, 2018-05-27:
Possibly relevant: https://github.com/cnr-isti-vclab/corto

pierotofy, 2022-07-07:
We could use https://github.com/DroneDB/libnexus or the novel `--3d-tiles` outputs for meshes.

pierotofy, 2022-10-28:
Also another option could be to display better compressed GLTF/GLB models. Most of the weight is due to the large textures which are mostly in PNG format. Some JPEG compression could go a long way.

smathermather, 2022-10-28:
Yes, switching to JPEG compression would fix 90%+ of these cases.

Saijin-Naib, 2022-10-28:
How about AVIF textures? We'd have very advanced compression that would outperform other lossy compression, support for arbitrary band numbers, and support for lossless alpha as well as optional lossless encoding. Plus, browser and OS support is good and it is open and actively improving constantly.

pierotofy, 2023-04-27:
Not enough bandwidth for this at the moment, plus GLB models now mitigate this issue quite well.

closed 2023-04-27 by pierotofy