WebODM/WebODM #385: Process exited with code 1 on DSM/DTM checked

merkato, 2018-02-01:
### How did you install WebODM? (Docker, natively, ...)?

Installed with docker.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 64 on Linux

### What is the problem?

Process exited with code 1 when DTM/DSM checked to process.

[INFO]    Initializing OpenDroneMap app - Sun Jan 28 08:34:01  2018
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/images
[DEBUG]   /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/images
[DEBUG]   Loaded kadluby_0017.JPG | camera: dji fc300c | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.31 | lat: 49.5491214167 | lon: 18.9208012222 | alt: 715.259
[DEBUG]   Loaded kadluby_0042.JPG | camera: dji fc300c | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.31 | lat: 49.5464505556 | lon: 18.9247374444 | alt: 714.959
(...)
[INFO]    Found 82 usable images
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[WARNING] Found a valid OpenSfM reconstruction file in: /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/opensfm/reconstruction.json
[WARNING] Found a valid Bundler file in: /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/opensfm/reconstruction.json
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[WARNING] Found a valid ODM Mesh file in: /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_meshing/odm_mesh.ply
[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[WARNING] Found a valid ODM Texture file in: /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_texturing/odm_textured_model.obj
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/coords.txt
[INFO]    Running georeferencing with generated coords file.
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/opensfm/bundle_r000.out -inputCoordFile /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/coords.txt -inputFile /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferencing_model_geo.txt
[DEBUG]   running ogr2ogr -overwrite -a_srs "+proj=utm +zone=34 +datum=WGS84 +units=m +no_defs" /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferenced_model.bounds.shp /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferenced_model.bounds.geojson
[WARNING] Found a valid georeferenced model in: /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferenced_model.ply
[INFO]    Running ODM Georeferencing Cell - Finished
[INFO]    Running ODM DEM Cell
[DEBUG]   running l2d_classify --help > /dev/null
[INFO]    Create DSM: True
[INFO]    Create DTM: True
[INFO]    DEM input file /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferenced_model.las found: True
[DEBUG]   running l2d_classify --slope 10 --cellsize 2  -o -s /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferenced_model.bounds.shp --outdir /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem --decimation 1  --initialDistance 0.15 /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing
[DEBUG]   running l2d_dems dsm /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem --slope 10 --cellsize 2  -o -s /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_georeferencing/odm_georeferenced_model.bounds.shp --outdir /var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem --maxsd 2.5 --maxangle 20 --resolution 0.1 --radius 0.1 0.3 0.9 2.7  --gapfill 
Creating ../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem/odm_georeferenced_model.bounds-0_dsm_r0.1 [idw] from 1 files
Completed ../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem/odm_georeferenced_model.bounds-0_dsm_r0.1 [idw] in 0:06:19.913330
Creating ../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem/odm_georeferenced_model.bounds-0_dsm_r0.3 [idw] from 1 files
Completed ../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem/odm_georeferenced_model.bounds-0_dsm_r0.3 [idw] in 0:06:36.056069
Creating ../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem/odm_georeferenced_model.bounds-0_dsm_r0.9 [idw] from 1 files
Completed ../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem/odm_georeferenced_model.bounds-0_dsm_r0.9 [idw] in 0:08:43.508101
Creating ../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem/odm_georeferenced_model.bounds-0_dsm_r2.7 [idw] from 1 files
Completed ../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem/odm_georeferenced_model.bounds-0_dsm_r2.7 [idw] in 0:28:22.807955
Completed gap-filling to create ../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem/odm_georeferenced_model.bounds-0_dsm.idw.tif in 1:11:20.318117
l2d_dems dsm completed (../var/www/data/21306b58-0506-497f-ac09-3dd049dec94a/odm_dem) in 2:01:26.025671

### What should be the expected behavior?

DEM, DTM and classified point cloud should be generated.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Using default parameters on small dataset (82 images), that is correctly processed with recent OpenDroneMap docker instance (created DSM/DTM), and correctly processing Meshing, texturing, orthophoto part on WebODM. Point cloud generated looks good.

pierotofy, 2018-02-01:
Seems related: http://community.opendronemap.org/t/error-while-processing-for-dtm/224

pierotofy, 2018-02-01:
I'm hesitant to investigate this issue further because the l2d module is going away with https://github.com/OpenDroneMap/OpenDroneMap/pull/743, so once that PR is merged the issue will likely be solved.

pierotofy, 2019-04-03:
This should be fixed.

closed 2019-04-03 by pierotofy