WebODM/WebODM #386: webodm.sh: line 22: source: .env: file not found

iqnaul, 2018-02-05:
### How did you install WebODM? (Docker, natively, ...)?

I installed using Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

I'm using google chrome and Windows 10 64-bit

### What is the problem?

After running bash: sh webodm.sh start I got this error:
webodm.sh: line 22: source: .env: file not found

Is there any recommendation for me?

Thanks

pierotofy, 2018-02-07:
Make sure you are starting from a fresh git copy of the repository. Perhaps the .env file was lost during a copy.

```
git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input
cd WebODM
./webodm.sh start
```

iqnaul, 2018-02-08:
okay, I deleted previous clone and did as you suggest @pierotofy but I got another error : 
"Error for db (2, 'WaitNamedPipe', 'The system cannot find the file specified.')
"Error for node-odm-1 (2, 'WaitNamedPipe','The system cannot find the file specified.')
"Windows named pipe error: The system cannot find the file specified. (code: 2)

Do I need to setup something?

Thanks

iqnaul, 2018-02-08:
Okay I have looked at #137 , Hyper-V and VT-X was already enabled, I also downgrade my Python version to 2.7.14. But there's another error come out:
...
Starting webapp --- done
No containers to start
Pulling db (opendronemap/webodm_db:latest)
Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)

iqnaul, 2018-02-17:
Hi @pierotofy I finally managed webODM worked! I reinstall docker again, and now it's working.

Thanks!

closed 2018-02-17 by iqnaul