WebODM/WebODM #387: http://localhost:8000 don't work after installation

gpsman, 2018-02-05:
--- REMOVE THIS LINE AFTER READING IT: Please open an issue only to report faults and bugs. For questions, comments and feature requests instead visit http://community.opendronemap.org/c/webodm. Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed.

### How did you install WebODM? (Docker, natively, ...)?

natively

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 58 on Ubuntu Linux (whatismybrowser.com/w/G7A46QX)

### What is the problem?

Firefox ne peut établir de connexion avec le serveur à l’adresse localhost:8000.
http://localhost:8000 don't work after installation (no connection)

### What should be the expected behavior?

working

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

It's a new installation (Ubuntu / VirtualBox) with last version
[Install WebODM.txt](https://github.com/OpenDroneMap/WebODM/files/1694499/Install.WebODM.txt)

Thanks

gpsman, 2018-02-05:
Additional information in attached file
[OperationalError at _.txt](https://github.com/OpenDroneMap/WebODM/files/1694560/OperationalError.at._.txt)

pierotofy, 2018-02-07:
Hi @gpsman try restarting your computer and issuing:

`./webodm.sh down && ./webodm.sh start`

gpsman, 2018-02-07:
Thanks, I vave tried but it's still the same.
It seems there is a docker error. I have tried pip uninstall docker-py; pip uninstall docker; pip install docker but without any result
[Error Docker connexion.txt](https://github.com/OpenDroneMap/WebODM/files/1704045/Error.Docker.connexion.txt)

bipulneupane, 2018-02-09:
./webodm.sh down && ./webodm.sh start
worked for me
thanks to @pierotofy

closed 2018-03-02 by pierotofy