WebODM/WebODM #388: Bluescreen on boot

Bengan1, 2018-02-06:
### How did you install WebODM? (Docker, natively, ...)?

Docker Toolbox

### What's your browser and operating system?

Chrome 63 on Windows 10

### What is the problem?

I installed it accodring to this guide: https://www.youtube.com/watch?v=kJQegRQUhWU 
When I tried to start up the system I hade an issue:

running pre-create checks...
Error wirh pre-create check: "Hyper-V is installed. VirtualBox won't boot a 64bits VM when Hyper-V is activated. It it's installed but deactivated, you can use --virtualbox-no-vtx-check to try anyways"
Look like something went wrong in step 'Checking if machine default exists'...
Press any key to continue...

After that I change and added the "--virtualbox-no-vtx-check" line.

Now I tried to startup, and start it via "Run as administrator". It start to boot, then efter 20 ish seconds the whole computer chrases and I get blue screen. 

### What should be the expected behavior?

It start to boot, then efter 20 ish seconds the whole computer chrases and I get blue screen.

pierotofy, 2018-02-07:
You shouldn't use Docker Toolbox if you have Hyper-V support. Try following the instructions listed here instead: https://github.com/OpenDroneMap/WebODM#getting-started

pierotofy, 2018-03-02:
If there are further questions about this subject,  according to the [guidelines](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md) please let's move them to the forum. http://community.opendronemap.org/c/webodm

closed 2018-03-02 by pierotofy