WebODM/WebODM #389: Access-Control-Allow-Origin header contains multiple values '*, *'

ligai322, 2018-02-06:
### How did you install WebODM? (Docker, natively, ...)?
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 63 on Windows 10

### What is the problem?

my website remote call webodm api ,  like  https://demo.webodm.org/api/projects/?format=json

my code:
[image]

the response:
[image]

error infomation:
[image]

pierotofy, 2018-02-07:
This looks like it's caused by both Django and nginx are adding their headers (duplicated in this instance).

This could likely be solved by modifying `/nginx/nginx.template.conf` by removing the appropriate header directives.

closed 2018-02-07 by pierotofy

pierotofy, 2018-02-07:
These changes will be reflected once we update the webodm_webapp image on docker hub, or when somebody rebuilds the image from scratch.

The demo site might continue to have the issue until next time we update it.

PR #390 (2018-02-07): Removed CORS headers from nginx, let django handle it
Fixes #389