WebODM/WebODM #393: Replace pooling with websockets

pierotofy, 2018-02-13:
This should be part of a large set of changes that allows node-odm to communicate with webodm by means of websockets instead of pooling. This would increase speed and performance.

Interesting links that could help: https://channels.readthedocs.io/en/latest/introduction.html

https://github.com/websockets/ws

pierotofy, 2018-05-22:
This is next as a precursor to a more scalable processing node infrastructure.

pierotofy, 2018-05-23:
After some thought, this might actually be pushed aside for the time being. While websockets would indeed allow for faster updates and scale vertically, they create a stateful environment which creates all sorts of complications and most importantly, will be an impediment toward horizontal scaling.

It might be interesting to apply websockets on the web browser client to read updates about the output console, maybe task status and a few others, but as far as node-odm <--> WebODM communication is concerned, the advantages might not be worth the effort (and might actually hinder horizontal scaling later on).

smathermather, 2018-05-23:
Make sense.

closed 2019-01-16 by pierotofy