WebODM/WebODM #396: very high browser memory consumption

tobi-kellner, 2018-02-17:
Hi,

I know this is probably not WebODM's "fault", but it's currently limiting my ability to use WebODM:
If I try to upload a large number of pictures, the browser tab crashes, most likely due to very large memory consumption.

### How did you install WebODM? (Docker, natively, ...)?

Docker running on Win 10

### What's your browser and operating system? 
Chrome 64 on Windows 10

### What is the problem?

Trying to upload a large project (600+ pics, each 4000x3000) with default resize settings, the browser tab crashes, I get the Chrome "Oh Snap..." error page.
With a smaller number of images (340pics, 1.59GB), I can upload them, but the browser tab consumes a huge amount of memory. The memory consumption of the WebODM Dashboard tab alone peaks at 2GB and remains at around 400 GB long after all uploads are completed.

### What should be the expected behavior?

The memory footprint of the browser should remain reasonable (especially as a lot of RAM is required by the VM to run ODM), and should definitely not limit the number of images that can be processed (as it does in my case).

### How can we reproduce this? 

Try and upload and resize a large number of images.

Thanks for looking into this!!

pierotofy, 2018-02-17:
Hi @tobi-kellner :hand: thanks for pointing this out. When I first introduced the in-browser resizing feature I tested it with ~3000 JPGs (but I also have a laptop with 24G of RAM, which is more than your average laptop).

In-browser resize is slow and consumes memory, but has the tradeoff that images stored on the server are already resized == less space and bandwidth.

Ideally we should offer a way to resize the images on the server (we once had that option, but has been since removed in OpenDroneMap core, now the --resize-to parameter simply instructs opensfm to resize the images for the purpose of computing the point cloud, texturing still happens at full image resolution which is desirable in terms of quality, but is also more memory intensive).

So the question of how to approach this is either:

1. Add resize server side on WebODM
2. Add resize server side on node-OpenDroneMap
3. (Re)add resize server side on OpenDroneMap (which is then reflected on node-OpenDroneMap)

I don't particularly like 2, but 1 and 3 sound reasonable.

What are your thoughts @dakotabenjamin ? I know we're removed the resize_images step several months ago and that opensfm already does its own resizing, some lower end computers however cannot handle full resolution texturing. Should this be added to WebODM only?

In the meanwhile the workaround is just skip the in-browser resize by changing the appropriate select box from "Yes" to "Skip", or resize the images via another program (look out for exif tags however).

tobi-kellner, 2018-02-17:
@pierotofy , thanks for considering the issue!
I guess this is only an issue for people like me who try to do a lot (several 100 pics) with a little (16 GB RAM, with WebODM running in a VM).

I am perfectly happy for this item to be very, very low down on your priority list, I guess I was kind of hoping against hope that there could be a simple way to get the browser to go about the task more efficiently (it seems to hold many/all images in memory at the same time rather than resizing & sending one at a time...)

For now, I am simply skipping resizing, hope that doesn't make me crash into the next bottleneck ;)

pierotofy, 2018-02-17:
I can confirm that the resizing code seems to be leaking memory (I notice an increase of ~30 MB for every 50 images), but at ~368 images memory on Chrome on Linux peaked at 777 MB (so quite a bit less than Windows). I wonder if you would have better results on Firefox.

[image]

Regardless, definitely something to fix / improve.

pierotofy, 2018-02-17:
Notice also the waste of CPU resources since everything runs on the UI thread.

Ideally we should offer both options (client resize or server resize), with the default being server resize.

dakotabenjamin, 2018-02-19:
@pierotofy let's look at the resize stuff once I've integrated the opensfm
georeferencing, because that's the main reason it was removed. We could use
opensfm's resized images through the rest of the toolchain, for instance.

pierotofy, 2018-03-07:
This is resolved as of latest master. We can now resize on both the server or browser (and we resize GCPs also!)

closed 2018-03-07 by pierotofy