WebODM/WebODM #397: Where is WebODM task/data stored?

iqnaul, 2018-02-17:
### How did you install WebODM? (Docker, natively, ...)?

I installed it using docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Google Chrome, Windows 10 Pro 64-bit

### What is the problem?

I tried to create my first task in WebODM, analyzing 13 images from my drone and succeed to have orthomosaic and surface model images. After doing this I realized that my harddrive capacity occupied +/- 5 GB. Where are the webodm database stored?

Because I want to backup it first.

Thanks

pierotofy, 2018-02-17:
Hi @iqnaul does this https://github.com/OpenDroneMap/WebODM#where-are-my-files-stored answer the question?

tobi-kellner, 2018-02-17:
Note, however, that it seems the` --media-dir` parameter only determines where  the uploaded files and the final products are stored.
During the processing, a large amount of disk space (many times the size of the images uploaded) is still stored inside the VM, in the Docker containers (??). You can use
`docker system df`
and
`docker system df -v`

to see what's happening.
After crashing my machine during processing, I once somehow managed to end up with a large amount of "stranded" data - stranded in the sense that it still took up space even though the task had been deleted (in its half-finished state) in WebODM.
A combination of 
`
./webodm.sh down
`
and
`
docker container prune
`
sorted it (do it at your own risk, though - I also once lost all my projects playing with this)...

iqnaul, 2018-02-20:
Hi @pierotofy after doing what @tobi-kellner I found that odm docker repo has 5 GB of data. How to delete this repo data? Because after I ran docker container prune and do docker system df -v it still showing the same data amount for each opendronemap repositories.

Any suggestion?

tobi-kellner, 2018-02-20:
@iqnaul , 
where are the 5GB? If they are in `Images` then I think that's simply how much memory WebODM needs.
I found that `Containers` is the one that fills up during processing and sometimes does not get cleared up if something crashed.
Processing up to around 300 pictures of 12Mpix size, my Docker (dynamic) disk's total size has peaked at around 35GB during processing. So it seems a good rule of thumb for me could be:
5GB for the programme
10GB for every 100 pictures, temporarily during processing

pierotofy, 2018-03-02:
If there are further questions about this subject,  according to the [guidelines](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md) please let's move them to the forum. http://community.opendronemap.org/c/webodm

closed 2018-03-02 by pierotofy