WebODM/WebODM #398: Suggestion: alternative visualisation for running tasks

tobi-kellner, 2018-02-17:
Hi,

this is not a big report or even a simple feature request but a general idea:

I think the web interface for running tasks could be improved, especially for those (like me) who are not familiar enough with the inner workings of ODM.
My basic suggestion is to illustrate the total process as a number of steps/stages, corresponding to the steps available for `--rerun` etc
For steps where progress can be determined from output, for example because a certain RegEx would match once for every picture processed, a progress bar for this step could be displayed.
Optionally, a sentence or two about what's happening in this step could be displayed, ideally even with info on what are the main parameters that influence this particular stage.

I am not sure how I could help with this, I guess I could produce some mock-ups of what the overall thing could look like...

This is super low-priority, just an idea I had from playing around with WebODM for a few days.

pierotofy, 2018-02-17:
Hi @tobi-kellner,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-02-20 16:40) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2018-02-17:
I like the idea of adding better, more friendly status indicators. The raw console output can be intimidating.

tobi-kellner, 2018-02-18:
@pierotofy , I'm afraid I don't know exactly how I can be of help with this, my HTML5/JS/DOM knowledge is very basic (I assume it's client-side JS/DOM stuff that would need to be done)?
Plus, I don't really know what parts of the raw console output correspond to what steps of the process.
I can produce a general mock-up of what I think it could look like, to get the ball going....

tobi-kellner, 2018-02-21:
@dakotabenjamin , @pierotofy ,
I could make some suggestions about changes to the UI for WebODM (about restructuring how parameters are presented, and what info is displayed during processing), but I would not be able to implement these changes myself. I could provide some mock-up screenshots, and perhaps some pseudo-code about how I'd implement if I knew anything about HTML5/JS/DOM...
My questions:
- is there a point in doing this, or is it unrealistic that anyone else would actually implement these?
- if there is a point, where would be the best place to post them? here?

Thanks,
Tobi

pierotofy, 2018-02-21:
It's not unrealistic, but we can't make any time guarantees on when things will get implemented either (unless funding for implementing a particular feature is provided). All feedback and suggestions are always appreciated.

So I'd say go ahead! Here is the right place to post them :+1:

pierotofy, 2018-12-10:
Completed as part of #571

closed 2018-12-10 by pierotofy