WebODM/WebODM #400: [QUESTION] How does WebODM determine the measurements with the 3D model?

FarhadG, 2018-02-21:
I see that WebODM has a measuring tool for determining distance and volumes of 3D models. To be able leverage/modify this sort of functionality, I'm curious how it's figuring out the scale of how many pixels equate to the number of feet/meters/etc. 

Any articles on this topic or reference points within the code to explain this would be much appreciated.

Thank you!

pierotofy, 2018-02-21:
All models are in meters. This is an implicit unit and its known before hand (since the point cloud output from OpenSfM is in meters, so are the models).

FarhadG, 2018-02-21:
Thank you for the response, @pierotofy. I guess more clearly, I'm trying to decipher how meters translate to the pixels on the screen. I see that there is a measuring tool that calculates the distance between two points (taking into account the camera projection, depth, etc.) -- how is this calculation made? Are you saying that the OBJ file containing the vertex (10, 8, 10) and (2, 4, 5) are actually X meters apart because these points are laid out using meters?

pierotofy, 2018-02-21:
You got it :)

(0 0 10) and (0 0 15) are 5 meters apart.

FarhadG, 2018-03-01:
Many thanks! Worked like a charm! :)

closed 2018-03-01 by FarhadG