WebODM/WebODM #402: odm_meshing core dump

bbranham3, 2018-02-21:
### How did you install WebODM? (Docker, natively, ...)?

Vmware esxi image of ubuntu, docker install 

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome,Windows 10 client,  Ubuntu server 

### What is the problem?

[DEBUG]   running /code/build/bin/odm_meshing -inputFile /var/www/data/0b48ca4c-25b0-47c7-b263-7b173abc4e7b/opensfm/depthmaps/merged.ply -outputFile /var/www/data/0b48ca4c-25b0-47c7-b263-7b173abc4e7b/odm_meshing/odm_mesh.ply -logFile /var/www/data/0b48ca4c-25b0-47c7-b263-7b173abc4e7b/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9  -samplesPerNode 1.0 -solverDivide 9
Illegal instruction (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_meshing.py", line 79, in process
    '-samplesPerNode {samples} -solverDivide {solver}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132

### What should be the expected behavior?

proper project completion

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

import of files to mesh

pierotofy, 2018-02-21:
I'm guessing this is a problem with the  CPU instruction set that Vmware esxi supports? Docker images work with CPUs with 64-bit extensions, MMX, SSE, SSE2, SSE3 and SSSE3 instruction set support or higher.

pierotofy, 2018-03-02:
If there are further questions about this subject,  according to the [guidelines](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md) please let's move them to the forum. http://community.opendronemap.org/c/webodm

closed 2018-03-02 by pierotofy