WebODM/WebODM #406: Plugin Repository

pierotofy, 2018-03-02:
It's not going to be sustainable to include all plugins in the WebODM repository. Ideally users should be able to publish plugins on separate GitHub repositories, then have a command line utility (or UI) able to browse/add/remove/enable/disable plugins, fetch them from the repository, etc.

We would probably keep a list of plugins in the WebODM repo.

pvgmenegasso, 2021-05-21:
Users could add links to their plugin repos into a file and them we could implement an interface to fetch them on the application and display it, right ?
I can help with that if that is possible

pierotofy, 2021-05-21:
Something like that; plugins already have a `manifest.json` file, so a repository would include a central location listing an index of all repository URLs, which needs to be parsed to update such index. Then WebODM could display the index and have the option to download plugins on-demand. We'd welcome contributions.

dlebauer, 2021-09-14:
@pierotofy was a plugin repository or list of plugins ever created? I checked the docs, google, forums and haven't found anything. 

It would be great to know which plugins are available, which are in development, and which might welcome contributions / collaboration.

pierotofy, 2021-09-14:
Nop, a plugin repository was never created. All plugins are still in https://github.com/OpenDroneMap/WebODM/tree/master/coreplugins. We always welcome contributions to improve any of them.

closed 2024-05-31 by pierotofy