WebODM/WebODM #415: failing on image processing

farmdude, 2018-03-23:
### How did you install WebODM? (Docker, natively, ...)?

I installed using Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 65 on Mac OS X 10

### What is the problem?

I processed 7 images of a 232 image set as a test run and they processed fine. When I went to run the full set 232 images, I keep getting errors. The only modification to the options was that I increased the min-num features to 16000 from 8000. It ran perfectly on the first 7 images, so I decided it should work on the rest.

### What should be the expected behavior?

It should process my images.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I allocated 16 gb ram (I am maxed out at 16gb) and 4 cpus to Docker before I ran the full set of images. The 7 image set had probably 12 GB of ram and 2 cpus allocated. The only thing I did was delete the 7 image task and then create a new task and add the 232 image set change the min-num features and start it. I tried to run this in both debug mode and production mode. both times I received errors. I am going to attach the terminal information as well. I'm not sure if it's helpful or contained in the output from the downloadable file in WebODM. I am currently uploading the image set. It is fairly large for my upload speed, so it may be a few hours before they are all there. The link to my Dropbox folder is: https://www.dropbox.com/sh/mksjw8sluccm0yf/AAARGklS4KeQI-8TgKFtxWnoa?dl=0

farmdude, 2018-03-23:
If it would work better to use OpenDroneMap INSTEAD of WebODM, I can install ubuntu on another drive on my computer and try that. I currently do not have the ability to increase my memory though. I thought about building a brand new computer to install ODM, but I don't have the funds for that currently. I first tried running ODM on a 10 year old computer and couldn't get it installed for some reason, so I thought maybe it would be better with WebODM. But it kept failing on the old machine. Figuring it was the machine, I switched over to my MacMini. Newer, but still a six year old model, but twice the memory (16GB and 10 times the CPU.)

pierotofy, 2018-03-23:
Hey @farmdude, what errors more specifically are you getting? Could you copy/paste the output of your task?

farmdude, 2018-03-23:
Thanks for your reply @pierotofy, the output is in my dropbox folder, but I will attach it here as well.
[task_output.txt](https://github.com/OpenDroneMap/WebODM/files/1842711/task_output.txt)

I reran the process after stopping everything and restarting. This time I used 115 images instead of the 232 and it said completed, but then it said an orthophoto could not be generated. I am attaching that task feedback here.
[noorthphotogenerated.txt](https://github.com/OpenDroneMap/WebODM/files/1842715/noorthphotogenerated.txt)

pierotofy, 2018-03-23:
Mm, the first problem seems a possible bug in OpenSfM. If you reprocess the dataset a second time, do you always get:

```
Aborted
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/run_opensfm.py", line 133, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
```

?

The error message about the orthophoto not being able to be generated seems odd, since from the log you pasted shows that an orthophoto was indeed generated. WebODM will display that message when an odm_orthophoto.tif file is not present.

Which of the 115 images did you use (the first 115)?

farmdude, 2018-03-23:
@pierotofy I'm not sure if it's the same error message every time. I deleted the logs after trying to process them. I used the first set of 115 images. I tried running again this afternoon with the first 62 images, and it succeeded. The orthophoto looks great of the vegetation, but there were some buildings and it really distorted the buildings.
[image]

If it helps I can try processing the larger set and see if I get the Aborted error that you referenced above. Thanks again for your help!

farmdude, 2018-03-23:
I just noticed the surface model is not present on the set of 62 images that I processed. I'm not sure why. I didn't see an error. I'll copy the output and being a novice, I'm not sure that I'll be able to tell, but I'll post it here.
[1:4imagescompletednosurface.txt](https://github.com/OpenDroneMap/WebODM/files/1843690/1.4imagescompletednosurface.txt)

farmdude, 2018-03-24:
Well, this is weird... I decided to try the first set of 115 images again. Changing absolutely no settings, the processing worked on the set of 115 images. Even though the set of 232 failed multiple times, and 115 failed originally... No idea, what would have caused it to work this time and not before. I did not restart docker or webodm between processing the 62 and 115, but I did restart everything when the processing was failing... I am going to try and process the full set today. and see what happens.

farmdude, 2018-03-24:
I decided to change the default resizing to something smaller, because for my purposes I don't really that large of an image. I changed the default resize way down to 648, and kept the min-num features set at 16000. I successfully processed 232 images sort of... When I viewed the final image, it's really distorted everywhere as opposed to the previous images that I had processed. I'm not sure what is causing this or why I'm having these issues...
[image]

farmdude, 2018-03-25:
So, is this problem because my computer doesn't have enough memory/new enough of a processor? My computer is a mid 2012 Mac Mini with 16 gb ram. If so, I may try and upgrade later this year. Also, if I had enough processing power and memory, would I need to resize the images before processing?

pierotofy, 2018-03-25:
I'm guessing the results look "weird" because you resized the images too much, while keeping a large `min-num-features` parameter.

Check the memory settings for your docker environment https://stackoverflow.com/questions/32834082/how-to-increase-docker-machine-memory-mac/39720010#39720010, 16 GB should be plenty to process 260 images at good resolution.

farmdude, 2018-04-10:
Also, the dsm/dtm output is not working for me for some reason... I really appreciate your feedback! This is a great program, I'm assuming I'm doing something wrong! I'm attaching the output task results.
[taskoutput.txt](https://github.com/OpenDroneMap/WebODM/files/1894811/taskoutput.txt)

[image]

pierotofy, 2018-04-10:
Color could be related to https://github.com/OpenDroneMap/OpenDroneMap/issues/801

pierotofy, 2020-04-02:
Please try to reprocess with the latest ODM/WebODM version. We've done a lot of improvements since to deal with the problems reported here. If the problems are still there, re-open a new issue in the ODM repo? :clinking_glasses:

closed 2020-04-02 by pierotofy