WebODM/WebODM #417: multiple "Access-Control-Allow-Origin: *" on response headers

foxsake, 2018-03-26:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 65 on Windows 10, Firefox 59 on Windows 10

### What is the problem?

2 "Access-Control-Allow-Origin: *" on response headers
error on console:
`Failed to load http://172.16.10.11:8000/api/token-auth/: The 'Access-Control-Allow-Origin' header contains multiple values '*, *', but only one is allowed. Origin 'http://localhost' is therefore not allowed access.`
response headers:
`HTTP/1.1 200 OK
Server: nginx/1.13.8
Date: Mon, 26 Mar 2018 08:31:28 GMT
Content-Type: application/json
Content-Length: 180
Connection: keep-alive
Allow: POST, OPTIONS
Vary: Accept, Accept-Language, Cookie
Content-Language: en
Access-Control-Allow-Origin: *
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, POST, OPTIONS
Access-Control-Allow-Headers: DNT,X-CustomHeader,Keep-Alive,User-Agent,X-Requested-With,If-Modified-Since,Cache-Control,Content-Type,Content-Range,Range
Access-Control-Expose-Headers: DNT,X-CustomHeader,Keep-Alive,User-Agent,X-Requested-With,If-Modified-Since,Cache-Control,Content-Type,Content-Range,Range`

### What should be the expected behavior?

should only be 1.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

running
```
$.ajax({
    type: "POST",
    crossDomain: true,
    url: 'http://172.16.10.11:8000/api/token-auth/',
    data: {'username': 'user', 'password': 'pass'},
    dataType: 'json',
    xhrFields: {
      'withCredentials': true
    },
    success: function(res) {
      webodmtoken = res['token']; 
      console.log(webodmtoken);
    }
  });
```
from a different domain..

similar to this issue
https://github.com/OpenDroneMap/WebODM/issues/389
but im already running the latest version and i also checked nginx config and the old CORS config was already removed.

pierotofy, 2018-03-26:
Mm, yes I can confirm we are writing the headers twice:

[image]

For some reason this seems to upset only Firefox (but works fine with Chrome). Will need to dig a bit to see if this is Django or Nginx's fault.

Thanks for reporting the problem!

pierotofy, 2018-03-31:
@foxsake could you try to update WebODM and try again? I've updated the Django cors headers module and run a test again and seemed to work fine. Perhaps this problem was caused by us upgrading to Django 2.

If you can verify that this works I'll close the issue.

foxsake, 2018-04-03:
@pierotofy it now works fine after update. Thanks!

closed 2018-04-03 by pierotofy