WebODM/WebODM #424: Development environment hangs up. Only shows warnings, no errors.

kdorosh, 2018-04-02:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 65 on Windows 10

### What is the problem?

Cloned a clean repo with `git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input`

`./webodm.sh start` works perfectly.

`./webodm.sh down` to switch to dev environment

`./devenv.sh start` runs and gives the following stack trace:

```
$ ./devenv.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Starting development environment...
Creating broker ... done
Creating worker ... done
Creating webapp ... done
Creating node-odm-1 ...
Creating broker ...
Creating worker ...
Creating webapp ...
Attaching to db, broker, node-odm-1, worker, webapp
db            | LOG:  database system was shut down at 2018-04-02 04:22:41 UTC
db            | LOG:  MultiXact member wraparound protections are now enabled
broker        | 1:C 02 Apr 04:26:03.121 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
broker        | 1:C 02 Apr 04:26:03.121 # Redis version=4.0.9, bits=64, commit=00000000, modified=0, pid=1, just started
broker        | 1:C 02 Apr 04:26:03.121 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
node-odm-1    | info: No tasks dump found
db            | LOG:  database system is ready to accept connections
worker        | Postgres is up - executing command
node-odm-1    | info: Checking for orphaned directories to be removed...
worker        | wait-for-it.sh: waiting 15 seconds for broker:6379
worker        | wait-for-it.sh: broker:6379 is available after 0 seconds
webapp        | Postgres is up - executing command
webapp        | wait-for-it.sh: waiting 15 seconds for broker:6379
webapp        | wait-for-it.sh: broker:6379 is available after 0 seconds
webapp        |
webapp        |  _       __     __    ____  ____  __  ___
webapp        | | |     / /__  / /_  / __ \/ __ \/  |/  /
webapp        | | | /| / / _ \/ __ \/ / / / / / / /|_/ /
webapp        | | |/ |/ /  __/ /_/ / /_/ / /_/ / /  / /
webapp        | |__/|__/\___/_.___/\____/_____/_/  /_/
webapp        |
webapp        |
webapp        | Checking python version... 3.x, good!
webapp        | Checking GDAL version... GDAL 2.2.4, released 2018/03/19, excellent!
webapp        | Setup git modules...
worker        | Checking for celery...   OK
worker        | Starting worker using broker at redis://broker
worker        | INFO Initializing GRASS engine using /usr/bin/grass74
worker        | /usr/local/lib/python3.6/site-packages/celery/platforms.py:795: RuntimeWarning: You're running the worker with superuser privileges: this is
worker        | absolutely not recommended!
worker        |
worker        | Please specify a different user using the -u option.
worker        |
worker        | User information: uid=0 euid=0 gid=0 egid=0
worker        |
worker        |   uid=uid, euid=euid, gid=gid, egid=egid,
webapp        | Setup npm dependencies...
webapp        | npm WARN deprecated babel-preset-es2015@6.24.1: ≡ƒÖî  Thanks for using Babel: we recommend using babel-preset-env now: please read babeljs.io/env to update!
webapp        | npm WARN deprecated tween.js@16.6.0: This package has moved to @tweenjs/tween.js. Please update your dependencies.
```

### What should be the expected behavior?

I should be able to access the webapp in my browser. The script should complete like `./webodm.sh start` and indicate that everything completed. Going to `http://192.168.99.100:8000` should show me the webapp like it does for `./webodm.sh start`.

I would also expect to see this somewhere in the stack trace output of `./devenv.sh start`:

```
webapp        | Congratulations! Γöö@(∩╜ÑΓùí∩╜Ñ)@ΓöÉ
webapp        | ==========================
webapp        |
webapp        | If there are no errors, WebODM should be up and running!
webapp        |
webapp        | Open a web browser and navigate to http://localhost:8000
```

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Cloned a clean repo with `git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input`

`./webodm.sh start` works perfectly.

`./webodm.sh down` to switch to dev environment

`./devenv.sh start` (see the stack trace provided above)

pierotofy, 2018-04-03:
I'll have to look into this. Thanks for reporting the problem.

pierotofy, 2018-04-15:
Hey @kdorosh, I can't seem to be able to reproduce this issue... have you found the cause of the problem? If you start from a fresh git clone and skip the `./webodm.sh start` command does it work?

Maybe this is Windows dependent (I'm testing from Arch Linux).

kdorosh, 2018-04-15:
Yes, from a fresh clone `./webodm.sh start` works.
From a fresh clone `./devenv.sh start` hangs up.
I imagine it's Windows dependent. It may be relevant that I'm using Docker Toolbox instead of Docker Community Edition.

pierotofy, 2018-04-15:
What location is the WebODM directory checked out to? Is it outside the C:\Users folder? By default Docker on Windows does not allow you to mount shares outside of your home folder (see http://support.divio.com/local-development/docker/how-to-use-a-directory-outside-cusers-with-docker-toolbox-on-windowsdocker-for-windows), and the dev environment does indeed mount a folder to allow changes to be placed.

pierotofy, 2018-04-16:
Are you running from C:\users\<youruser>\WebODM ?

kdorosh, 2018-04-16:
C:\Users\Kevin\git\WebODM -- I am

pierotofy, 2018-04-16:
I wonder if this is a problem related to npm... npm is notorious for creating very long paths that sometimes trip up NTFS. I can probably look into this more next week when I get access to a Windows machine.

pierotofy, 2020-04-02:
This should have been fixed (I last tested the --dev flag on a Windows 10 machine a few months ago).

closed 2020-04-02 by pierotofy