WebODM/WebODM #426: Could not find ccd_width in file. Use --force-ccd or ...

paulgodard, 2018-04-04:
Hi
I have installed Docker for Mac v18.03.0.  
Then in Terminal I typed : 
git clone https://github.com/OpenDroneMap/OpenDroneMap.git
then cd OpenDroneMap
docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_texturing:/code/odm_texturing opendronemap/opendronemap
Unable to find image 'opendronemap/opendronemap:latest' locally
latest: Pulling from opendronemap/opendronemap
281a73dee007: Pull complete 
2aea1b77cff7: Pull complete 
59a714b7d8bf: Pull complete 
0218064da0a9: Pull complete 
ebac621dcea3: Pull complete 
a3ed95caeb02: Pull complete 
b580731643cc: Pull complete 
faa5fbdba239: Pull complete 
a3480ec159f7: Pull complete 
Digest: sha256:76d12a83f5991f08980aac6bf6bc6dc535308a6fd37a2aa066d71269c725c579
Status: Downloaded newer image for opendronemap/opendronemap:latest
[INFO]    Initializing OpenDroneMap app - Wed Apr 04 10:18:07  2018
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /code/images
[DEBUG]   /code/images
[ERROR]   Not enough supported images in /code/images
[INFO]    OpenDroneMap app finished - Wed Apr 04 10:18:07  2018

Then I put 420 jpg files into the images folder and run the same code once more
docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_texturing:/code/odm_texturing opendronemap/opendronemap
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0249.JPG | camera: dji fc6310 | dimensions: 4864 x 3648 | focal: 8.8 | ccd: None | lat: -22.7970856944 | lon: 17.1284855556 | alt: 1903.383
[INFO]    Found 420 usable images
[DEBUG]   Line: WGS84 UTM 33S
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    OpenDroneMap app finished - Wed Apr 04 15:50:23  2018

Then I added the flag force-ccd
docker run -it --rm --force_ccd -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_texturing:/code/odm_texturing opendronemap/opendronemap
unknown flag: --force_ccd

Am I doing something wrong?
What is the size of the P4P sensor? Where do I put that value?

Help please, Paul

pierotofy, 2018-04-04:
Hi @paulgodard,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-04-07 12:05) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

paulgodard, 2018-04-04:
I believe that the correct place should be the forum instead... sorry!

closed 2018-04-04 by paulgodard