WebODM/WebODM #429: Reference plane for volume calculation

taoansf, 2018-04-10:
Hello,

I'm a new user of WebODM and I would like to calculate the volume of a hole in my surface model. My problem is that I can't find an option to select if the volume is calculated below or above my reference plane. I'm supposing that WebODM uses the evelation of the boundaries of my calculation polygon as the reference plane?

This site explain better my question
http://desktop.arcgis.com/en/arcmap/10.3/tools/3d-analyst-toolbox/surface-volume.htm

I think that option could be very useful when you need to calculate the volume in different situations. For example, calculate the volume that fill a hole (below the reference plane) or a volume of a hill (above my reference plane).

Thank you for any help.

With best regards,

Taoan

pierotofy, 2018-04-10:
Hi @taoansf,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-04-13 14:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2018-04-10:
@taoansf the reference plane is interpolated from the polygon boundaries (only from the points you click on the map). Then the reference plane is subtracted from the digital surface model of the area. We always return the absolute value of the difference between the reference plane and the DSM, so it shouldn't matter whether you are measuring a hole or a volume. 

The full script is available here: https://github.com/OpenDroneMap/WebODM/blob/master/plugins/measure/calc_volume.grass

taoansf, 2018-04-11:
thank you, got it
cheers!

closed 2018-04-11 by taoansf