WebODM/WebODM #430: Why does the process hang on a random image during reconstruction?

bluehubbardsquash, 2018-04-13:
### How did you install WebODM? (Docker, natively, ...)?

Via an .exe

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 59 on Windows 10pro

### What is the problem?

For a number of tasks executed the progress window does not update. It gives the appearance that the task hung on processing an image. However, the cog icon continues to spin and it is noted nearby that the task is "running". After I close down WODM and shutdown my computer, when I log back into WODM, the task indicates it's now complete. The progress window output also appears complete and indicates that the task is done. I'm also able to download the imagery products.

### What should be the expected behavior?

The task progress window should continue to provide updates as to subtasks which were complete and when the task is finished. The cog icon should stop spinning and it should say "complete" beside the cog. I should be able to access/download the WODM products. This should happen without me having to close out of WODM and shutdown my computer.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I set up tasks to run a mosaic. No other options are selected except the min-num features is set to 20,000. I have a HyperV machine allocated with 25gb ram. I received this error when running tasks with as little as 200 images.
This is the reference link to the ODM forum:
http://community.opendronemap.org/t/why-does-the-process-hang-on-a-random-image-during-reconstruction/424
This is a link to the last set of imagery I ran when the error occurred:
https://1drv.ms/u/s!AsDrAVdsoyVVuyiWG_WhW85GEmq4

pierotofy, 2018-04-13:
This might have been fixed as part of #428. Could you try to update your version of WebODM and see if the problem happens again?

pierotofy, 2018-04-18:
This could be related: http://community.opendronemap.org/t/repeated-code-1-error-on-exit/486/7

pierotofy, 2020-04-02:
Closing since this should have been fixed. It if hasn't, please re-open. :clinking_glasses:

closed 2020-04-02 by pierotofy