WebODM/WebODM #431: Unable to install on fresh Ubuntu using script.sh

sifukd, 2018-04-13:
I've created a fresh ubuntu 16.04 VM to run natively. Ultimately i want to create a production machine that has loads of ram and ssd's.
i download the script file and run command:
./ubuntu_1604_install.sh and get error
./ubuntu_1604_install.sh line 7: syntax error near unexpected token `newline'
./ubuntu_1604_install.sh line 7: `<!DOCTYPE html>'

What could be the cause of the error?

pierotofy, 2018-04-13:
Hi @sifukd,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-04-16 09:10) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

sifukd, 2018-04-13:
Please ignore this, i right clicked and saved the .sh file from the webodm main page. It is actually a link to where to get the script. (should be better stated on website).
Jeremy

closed 2018-04-13 by pierotofy