WebODM/WebODM #434: Could not upload all files. An error occurred. Please try again.

smathermather, 2018-04-15:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system?

Chrome 65 on Windows 10 is what it reveals as, but really it's Brave.

### What is the problem?

Upload of images to WebODM fails with the message "Could not upload all files. An error occurred. Please try again."

### What should be the expected behavior?

Upload of files, and start of processing.

pierotofy, 2018-04-15:
Does it work on other browsers (Firefox)? Do you get any error messages if you open the web console in Brave?

smathermather, 2018-04-15:
As far as I can tell, Brave doesn't have a web console. That doesn't seem possible, but I can't find it.

pierotofy, 2018-04-16:
@wondo2 if you open the Chrome console (press F12) and go to the Console tab, do you see any error messages?

pierotofy, 2018-04-16:
Will look into it, thanks.

pierotofy, 2018-04-17:
@smathermather do you get the same problem is you use Firefox or Chrome?

smathermather, 2018-04-17:
I haven't had the issue with Firefox or Chrome.

DelphiR, 2018-05-14:
I got such type of errors several times but it was during dowloading for me.
So I introduced in the bug and see that the sometimes attempt to download a big file set (all asset) cuase to get more 2Gb of Ram to task. So the kernel may kill the task. so after that I get the 500 error

pierotofy, 2020-04-02:
WebODM now implements chunked uploads, which should have fixed these errors. If there are still problems, please re-open. :clinking_glasses:

closed 2020-04-02 by pierotofy