WebODM/WebODM #438: Task output shows duplicate lines

pierotofy, 2018-04-18:
For some reason the task output shows duplicate lines. Ex:

```
d105f50a7dc9/odm_georeferencing/odm_georeferencing_model_geo.txt[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/bundle_r000.out -inputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/geocoords_transformation.txt -inputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_model_geo.txt[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/bundle_r000.out -inputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/geocoords_transformation.txt -inputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_model_geo.txt[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/bundle_r000.out -inputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/geocoords_transformation.txt -inputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_model_geo.txt[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/bundle_r000.out -inputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/geocoords_transformation.txt -inputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_model_geo.txt[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/bundle_r000.out -inputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/geocoords_transformation.txt -inputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_model_geo.txt[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/bundle_r000.out -inputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/geocoords_transformation.txt -inputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /var/www/data/7b0360fc-4961-4fec-a9b6-d105f50a7dc9/odm_georeferencing/odm_georeferencing_model_geo.txt
```

closed 2018-04-19 by pierotofy

smathermather, 2018-04-19:
Will the next bug be related to the difficulty of naming things?

pierotofy, 2018-04-19:
Yes! lol

PR #440 (2018-04-19): Fixed off-by-one error, newlines handling
Fixes #438