WebODM/WebODM #439: Show Cameras and Pictures

pierotofy, 2018-04-19:
This is a feature that keeps being asked for. It would be cool to show where the camera shots are, with the ability to pull a thumbnail of the original image. Additionally it would be cool to show the coverage area for each camera.

smathermather, 2018-04-19:
That would rock, for sure.

saricicekibrahim, 2019-01-04:
Hi there,

On several photogrammetry software before processing only points of photos taken can be seen.
So as to calculate field of view, some more data near exif needed I think.
If image was taken 90 deg. camera angle, height of drone is needed. And it's not stored on exif.
Some of reference about that.
https://www.nrcan.gc.ca/earth-sciences/geomatics/satellite-imagery-air-photos/air-photos/about-aerial-photography/9687
https://photo.stackexchange.com/questions/56596/how-do-i-calculate-the-ground-footprint-of-an-aerial-camera
Heiht of dorne & camera angle can be taken from ui.
But than gps direction is needed, for exact bbox. And couldn't see a tag also on exif for that.
Any comment?

Anyway,

By selecting or drag & dropping images (which has gps detail in exif)
- location of shots, names, size can be seen on a list & map below;
https://saricicekibrahim.github.io/ImageLocationOnMap

I couldn't decide the scenario to use this kind of interface.
Instead of upload images button this may be used.
Uploading just selected images may also be added.
Does users need those points after process? May be points should be stored on project.

Bla bla bla...

0x906, 2019-01-04:
I might missed something but the field of view is a “property” of the camera. You don’t calculate it. You just go get the FOV angle from the manufacturer spec file.
Your second reference is about calculating the image footprint. If you are trying to do that, then you also need as well the CMOS geometry along with the flying height and focal length of the camera. Just remember that GPS is reporting the ellipsoid height as default and not the above ground flying height.

pierotofy, 2020-05-21:
Done! https://community.opendronemap.org/t/just-released-cameras-display/4422

PR #863 (2020-05-21): Adds support for displaying cameras
This PR aims to add support for displaying cameras.

[image]

Should close #439 #312
closed 2020-05-21 by pierotofy