WebODM/WebODM #443: Unable to upload Files

hubertokf, 2018-05-02:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 66 on Linux - Linux Mint 18

### What is the problem?

Messsage "Could not upload all files. An error occurred. Please try again." by the end of uploading files.

### What should be the expected behavior?

Finish files upload and start processing it

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

1. Started WebODM app using Docker. I have a nginx proxy (with SSL) to redirect to specific port. (using no-ip domain)
2. Created an account
3. Started a new project
4. Selected a dataset([Boruszyn dataset link](https://github.com/merkato/odm_boruszyn_kap))
5. Using the default parameters, started the upload
6. By the end of upload, the message appears: "Could not upload all files. An error occurred. Please try again."

pierotofy, 2018-05-04:
Hey @hubertokf perhaps this has something to do with your nginx proxy configuration. I think we should move the conversation over to http://community.opendronemap.org since it's probably not an issue with WebODM but a configuration problem specific to your environment. :+1:

closed 2018-05-04 by pierotofy

hubertokf, 2018-05-08:
Just to make document it. To solve the problem, I just set the `client_max_body_size;` to the nginx server configuration.

Thanks!